cjcliffe/CubicSDR #921: Bookmarks / labels broken in latest 0.25?

LouisJB, 2021-10-15:
Since upgrading to the latest I think channel labelling and bookmarks is a bit broken.
If you try to use it, create a new band and label it and bookmark it, or label any existing bookmark or band I think you will quickly see what I mean.

sageninjahawk, 2021-11-07:
I am able to confirm this issue:
When trying to add a new label or bookmark the program freezes.
Ran from the Terminal and this is the error that shows up:

(CubicSDR:136080): Gtk-CRITICAL **: 20:27:05.515: IA__gtk_widget_event: assertion 'WIDGET_REALIZED_FOR_EVENT (widget, event)' failed

edarnoldnet, 2022-06-13:
CANNOT add labels in 2.4. Won't even read them from XML file if formatted properly.

cjcliffe, 2026-11-14:
fixed in master and will be in 0.3.0 builds

closed 2026-11-14 by cjcliffe