cjcliffe/CubicSDR #926: error: ‘modem’ has not been declared

emanuele-dedonatis, 2021-11-04:
I'm not able to install CubicSDR for Ubuntu 20.04.

I followed all the steps from the wiki but I'm stuck on building CubicSDR
[image]

I'm supposing is a problem with wxWidgets: I tried with 3.1.5 and 3.1.3 but can't figure out what the problem is!

emanuele-dedonatis, 2021-11-04:
Found the problem: [master branch](https://github.com/jgaeddert/liquid-dsp) of **liquid-dsp** doesn't have modem definition anymore!

I built and installed **liquid-dsp** from [v1.3.2 tags](https://github.com/jgaeddert/liquid-dsp/tree/v1.3.2) and it works!

Please update the wiki!

nmaster2042, 2021-11-05:
I confirm building liquid-dsp from v1.3.2 tag is working.

ElwinCabrera, 2021-11-08:
I too can confirm that this build works when building liquid-dsp from v1.3.2 tag.

vsonnier, 2021-11-12:
The problem were with the Digital Modems only, where `modem` in liquid-dsp was renamed to `modemcf` in the current master. I think it is fixed now.

commit 0831aa26 (2021-11-12): Fix #926: liquid-dsp master 'modem' changed to 'modemcf' for Digital modems, updates Windows liquid-dsp binaries to master
- Updated Windows bins dependencies to latest MSYS2
- Win32/64 bins liquid-dsp ref: 471f46660ddfab7e6713341324f473b7785da101
- Win32/64 bins using GCC 11.2 (MSYS2)
closed 2021-11-12 by vsonnier