cjcliffe/CubicSDR #928: Building CubicSDR with Debian bulseye, CubicSDr segfaults

Foxone-1, 2021-11-22:
Debian Bullseye

xwWidgets 3.13. or 3.15 run CubicSDr from x64 segaults with either xwWidget library was working with Debian 9.

Nov 22 18:56:51  kernel: [ 5129.576455] CubicSDR[76807]: segfault at 21 ip 00007f28b535e2ff sp 00007ffec967e0c0 error 4 in libX11.so.6.4.0[7f28b5340000+8c000]
Nov 22 18:56:51 kernel: [ 5129.576460] Code: 00 48 89 f7 e8 92 27 fe ff 49 89 c0 31 c0 49 81 f8 fe ff 00 00 0f 87 a6 00 00 00 48 8b 85 68 09 00 00 48 85 c0 74 05 48 89 ef <ff> 10 ba 08 00 00 00 be 62 00 00 00 48 89 ef e8 8d 34 fe ff 4c 89
sdray@nostromo:~/CubicSDR/build$ cd ..

All steps followed from instructions

Foxone-1, 2021-11-29:
I appear to have managed to compile by using the latest wxWidget libs by following the instructions on https://www.wxwidgets.org/downloads/ and using the bulseye build. I then had to change the DwxWidgets to -DwxWidgets_CONFIG_EXECUTABLE=/u
sr/bin/wx-config which is where the package creates it rather than the default defined in the instructions.

This works but I do see some odd behaviour with screen layout but it works

vsonnier, 2021-12-12:
Great ! Have fun with CubicSDR !

closed 2021-12-12 by vsonnier