cjcliffe/CubicSDR #929: Appimage error LIBTHAI_0.1.25' not found

Twilight-Logic, 2021-12-02:
I have just downloaded the latest appimage file: CubicSDR-0.2.4-x86_64.AppImage

Unfortunately it will not run on Linux Mint 20.2 when I double-click the icon in the GUI. I get the following error when I launch it manually from the command line:

`./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libthai.so.0: version `LIBTHAI_0.1.25' not found (required by /lib/x86_64-linux-gnu/libpango-1.0.so.0)`

I thought that the appimage was self-contained anyway, although should the program be depending on support for a specific language in order to run it?

Twilight-Logic, 2021-12-02:
Ok, just realised that I didn't have one of the SDRplay libraries installed. This not done, I now get the following:

```
SDR enumerator starting.
SoapySDR init..
	API Version: v0.7.0
	ABI Version: v0.7
	Install root: ./////////
	Loading modules... 
[ERROR] SoapySDR::loadModule(.//////////lib/SoapySDR/modules0.7/libsdrPlaySupport.so)
  dlopen() failed: libmirsdrapi-rsp.so.2.13: cannot open shared object file: No such file or directory
	Available factories...airspy, airspyhf, audio, bladerf, hackrf, lime, null, redpitaya, remote, rtlsdr
./CubicSDR-0.2.5-x86_64.AppImage: symbol lookup error: /lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: xcb_dri3_get_supported_modifiers
```

I did find a previous issue that suggested opening launching CubicSDR using:

`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 ./CubicSDR-0.2.5-x86_64.AppImage`

This does launch the GUI, but CubicSDR does not see my RSP2 device.

Also in the startup I see:

```
Make device 1
[INFO] Make connection: 'DigiGreen [USB]'
[ERROR] libusb_open failed
[ERROR] Failed to open device
[ERROR] connection is not open
[ERROR] Write(64 bytes) failed
[ERROR] connection is not open
[ERROR] Write(64 bytes) failed
[ERROR] Failed to open. Device is busy.
Error making device: Failed to make connection with 'DigiGreen, media=USB, module=FX3, addr=34323:1204'
```
No idea what DigiGreen is, but that's not what I am trying to connect with and not something I have.

Just for completeness, SoapySDRUtil --probe="driver=sdrplay" reports:

```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=sdrplay
[INFO] devIdx: 0
[INFO] hwVer: 2
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSP2
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=2

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * ExtRef Enable - External Reference Control
       [key=extref_ctrl, default=true, type=bool]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Antenna A, Antenna B, Hi-Z
  Corrections: DC removal
  Full gain range: [0, 47] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 8] dB
  Full freq range: [0.001, 2000] MHz
    RF freq range: [0.001, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.096, 0.125, 0.192, 0.25, ..., 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```

Twilight-Logic, 2021-12-04:
I will just add that I have been able to compile from source and that seems to work OK.

vsonnier, 2022-01-15:
Great @Twilight-Logic ! With the diversity of Linux setups out there, the solution is often to rebuild from scratch.  You also have hhe advantage of using Cubic `master` compared to a severly outdated version (``0.24`) for which no support can be offered anyway.

closed 2022-01-15 by vsonnier

Twilight-Logic, 2022-01-15:
Version 0.2.4 appears to be being still being offered as the most current on cubicsdr.com?

https://cubicsdr.com/

The 'Downloads' button doesn't actually seem to do anything in my browser (Chrome or FF).
And linked to here:

https://github.com/cjcliffe/CubicSDR/releases/tag/0.2.4

This is where I downloaded the appimage from.

I have this morning stumbled on version 0.2.5:

https://github.com/cjcliffe/CubicSDR/releases

However, this was more by accident than by design and even that appears to be dated 2018. 

Please note that since I have an RSPduo, I started from the link on the SDRplay website, which took me to cubicsdr.com:

https://www.sdrplay.com/dlfinishs/

If this is incorrect, please let me know and I will be more than happy to inform SDRplay so that they can update their link.

vsonnier, 2022-01-15:
@Twilight-Logic  The author knows that his site [CubicSDR](https://cubicsdr.com/) is severly outdated, but somehow never had time to update it.

The Gihub Release page refers to the most recently tagged version, also severly outdated. I'm in the process of tagging a new one, at last : #937.  In the meantime, building the `master` is always the best solution.

Please note however, that I won't provide any Linux, or MacOS binaries for that matter. 
as I said in #937, I __may__ provide a Win64 `zip` as installer, but no more.

> Please note that since I have an RSPduo, I started from the link on the SDRplay website, which took me to cubicsdr.com:
https://www.sdrplay.com/dlfinishs/

ping @SDRPlay ?

Twilight-Logic, 2022-01-15:
Ok, noted.

Since I recently upgraded to the RSPduo, I just tried version 0.2.6a with this radio and it seems to work, at least in single tuner mode, but there is no audio. I see that there is some reference to setting up a 'Modem' in previous comments, but clicking Modem Setting seems to do nothing.  Given that you have just mentioned a new version (0.2.7) I will update the sources, re-compile and try again. Thanks.

Twilight-Logic, 2022-01-15:
Have pulled the latest sources from GitHub and tested. I has problems with launching the application and discovered that it must be started with the antenna set to Tuner 1 otherwise it hangs the SoapySDR/SoapySDRPlay stack. I can then switch to antenna 2 within the application (Settings => Antenna) and that works fine. 

With that sorted I can get into the application GUI but still no audio. The application does not show as an audio source in PavuControl/PulseAudio. Modem Settings contain no information and clicking on it still does nothing. Any help with resolving this would be appreciated.

[image]

vsonnier, 2022-01-15:
For the RSPDuo, you need API v3 and use the latest SoapySDRPlay built from source : https://github.com/pothosware/SoapySDRPlay3

Twilight-Logic, 2022-01-15:
I checked both SoapySDR and the SoapySDRPlay library versions before posting but forgot to mention it. Yes, they do need to be version 3 and my SoapySDR library was up-to-date. SoapySDRPlay had just one change and so was re-compiled and the updated version installed.

This in itself did not change anything, but clicking on the waterfall did! That brought the radio to life and the "Modem Settings" label now shows "FM Settings". An audio output device is now listed and I do get audio. I was not aware that the waterfall had to be clicked in order to start the VFO. As it is, one simply sees the waterfall running and no audio and wonders what is going on. It doesn't always start on the first click and sometimes a second click is required but it does work. Hopefully, your last comment will help others who run into the same problem. Thank you for your prompt replies and resolving that.

The application still needs to be started with the antenna set to Tuner 1, otherwise the next time it is started it just sits there waiting and showing  'Loading SoapySDR modules...". I can only assume that the SoapySDR stack has hung in some way and unfortunately so far I have found that the only way to clear this is to restart the OS. A log off and back on again is not sufficient and I don't see any process hung in the ps -ef output. Disconnecting and after a pause re-connecting the RSP does not help either.

Here are the errors I have seen in the console when this happens:

First time after launching with antenna set to Tuner 2:

Application termination complete.
Unhandled unknown exception; terminating the application.

Subsequently launching with antenna set to Tuner 2:

SDR enumerator starting.
SoapySDR init..
	API Version: v0.8.0
	ABI Version: v0.8
	Install root: /usr/local
	Loading modules... 
	Available factories...null, sdrplay
Available vertical sync SwapInterval functions: 
	glxSwapIntervalEXT: Yes
	DRI2SwapInterval: No
	glxSwapIntervalMESA: Yes
	glxSwapIntervalSGI: Yes
Using glxSwapIntervalEXT.

Loaded font 'Bitstream Vera Sans Mono' from '/home/johnc/src/CubicSDR/build/x64/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/johnc/src/CubicSDR/build/x64/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/johnc/src/CubicSDR/build/x64/fonts/vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/johnc/src/CubicSDR/build/x64/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/home/johnc/src/CubicSDR/build/x64/fonts/vera_sans_mono27_0.png', parsed 255 characters.
Terminating SDR thread..

ERROR: thread '9SDRThread' has not terminated in time ! (> 3000 ms)

vsonnier, 2022-01-15:
For Cubic general operation, you can either use the tooltips, or read the manual:  https://cubicsdr.readthedocs.io/en/latest/ even if it is a bit outdated. 

The remaining problems now lies with the SoapySDRPlay `module`, and you should direct questions there : https://github.com/pothosware/SoapySDRPlay3. 

Alas, the RSPDuo is the most tricky to use for CubicSDR because either Cubic or the `SoapySDR` abstraction do not understand well the distinction of "Master / Slave" and despite @fventuri incredible job there are still limitations and quirks compared to using SDRUno from SDRPlay.

fventuri, 2022-01-15:
In 'Single Tuner' mode the RSPduo should be able to switch antenna (which means switch tuner) without problems (it has been a bit since I last tried, so hopefully things are still working).

Also instead of rebooting the OS altogether, I would try to just restart the `sdrplay_api` daemon (using `systemctl restart sdrplay`) to see if it helps.
Sometimes you may have to stop it (`systemctl stop  sdrplay`), run a `ps` and `kill -9` the process, remove the shared memory files (`rm -f /dev/shm/Glbl\\sdrSrv*`), and finally restart the SDRplay API service (`systemctl start sdrplay`).

Franco

Twilight-Logic, 2022-01-16:
fventuri, thank you for that suggestion. 'systemctl restart sdrplay' does indeed appear to be sufficient to restart the SoapySDRPlay API layer. CubicSDR is then able to scan for the radio again and I can then select Tuner1.

Thank you for the hard work in making it possible for Cubic SDR to integrate with RSP SDR radios.

As to why it hangs the SoapySDR API I do not know. If I start by selecting Antenna Tuner1 from the start-up device detection screen then it works fine so this is my current workaround. Once the application is launched, only then do I switch to Tuner 2. However, upon next launch CubicSDR remembers the previous setting and if I forget to change it back to Tuner 1 then this is when the application crashes. The next time it is launched the API is hung. I can at least clear that now using your suggestion. If there is anything I can do to help troubleshoot this, please let me know.

I don't get this issue with SDRuno, but that is a different application and running on Windows rather than Linux so is not really a fair comparison.

I am not  sure what the Dual mode or master modes do. Having experimented a bit I find that there is no antenna selection. The application seems to run fine although I just get a very wide waterfall trail and a hiss so not sure quite what is going on there.

When running in Single Tuner mode it works beautifully and seems to be stable.

fventuri, 2022-01-16:
@Twilight-Logic - if you don't mind, please take a look at the config file `$HOME/.CubicSDR/config.xml`; there should be a `<device>` section there for your RSPduo, and in that section you should find something more or less like this:

```
            <antenna>Tuner 1 50 ohm</antenna>
```

I am interested to know what you have there after you exit from CubicSDR when you are using the second antenna/tuner.

As per the various modes that the RSPduo can run in, 'Single Tuner' is the standard SDR receiver like the RSP1A or RSPdx, while 'Dual Tuner' and 'Master/Slave' mode take advantage of the fact that the RSPduo has actually two semi-independent receivers: in 'Dual Tuner' mode you can use them with the same client application for things like diversity reception (but I don't think you can do that with CubicSDR), while in 'Master/Slave' you can use them in two different client applications, for instance CubicSDR and ADS1090, or even two instances of CubicSDR.

Franco

Twilight-Logic, 2022-01-17:
There are actually 4 <device> sections relating to the RSP Duo in config.xml. 

<id>SDRplay Dev0 RSPduo 2007075B32 - Single Tuner</id>
<id>SDRplay Dev1 RSPduo 2007075B32 - Dual Tuner</id>
<id>SDRplay Dev2 RSPduo 2007075B32 - Master</id>
<id>SDRplay Dev3 RSPduo 2007075B32 - Master (RSPduo sample rate=8Mhz)</id>

Focusing on the Single Tuner section the Antenna is set as follows:

<antenna>Tuner 2 50 ohm</antenna>

I also checked after forcing the API crash and it is still set the same.

Thank you for the brief explanation of the Dual Tuner modes. The dual application scenario you mention is what I would like to be able to do eventually.

fventuri, 2022-01-20:
@Twilight-Logic - I was able to reproduce this crash using the 'Tuner 2 50 ohm' antenna here, and I think this bug is due to the fact that the method `selectDevice()` was trying to re-select the RSPduo before it was properly initialized in the case of tuner 2 (which gets set via the `setAntenna()` method).

I made a change to the code of the method `selectDevice()` in the `SoapySDRPlay3` module, ran a quick test here, and it ran for a while without crashing.

I just pushed this change to the new branch `fix_rspduo_crash` (https://github.com/pothosware/SoapySDRPlay3/tree/fix_rspduo_crash); you can try it out by downloading the `SoapySDRPlay3` module from that branch, building it and installing it as per the instructions in SoapySDRPlay3 `README` file. 

If it also works for you (you may also want to try it with the tuner 1 antenna to make sure it works in that case too), I'll go ahead and push this code change to the `master ` branch.

Franco

Twilight-Logic, 2022-01-24:
Franco, that seems to work. It no longer hangs when starting with Tuner 2. Switching between Tuner 1 and Tuner 2 seems to work fine now.
Thank you.

fventuri, 2022-01-24:
@Twilight-Logic - glad to hear it worked for you too.

I just merged that code change into the `master` branch.

Franco