cjcliffe/CubicSDR #930: undefined reference to `modemcf_get_demodulator_evm'

vampjaz, 2021-12-10:
i've tried installing Cubicsdr from the AUR on my arch linux machine, and it failed in a number of different ways. so then i tried installing from scratch using the linux build instructions. during the linking stage, i get this error:

```
 48%] Linking CXX executable x64/CubicSDR
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/modules/modem/ModemDigital.cpp.o: in function `ModemDigital::updateDemodulatorLock(modemcf_s*, float)':
ModemDigital.cpp:(.text+0xde): undefined reference to `modemcf_get_demodulator_evm'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1461: x64/CubicSDR] Error 1
```

all the main dependencies have been compiled from scratch from a head branch as per the guide, and i've tried multiple different commits of cubicsdr with no luck. let me know if you want to see more of the logs.

vsonnier, 2021-12-11:
Duplicate of #926 : you have to use liquid-dsp `master` for Cubic due to recent changes that broke the previous API.

vsonnier, 2021-12-14:
OK, this *may* be a different issue than #926. Is it a compiler-specific problem on Linux? 

I have no problem building Cubic `master` and liquid-dsp `master` for Windows.

CristianRossiSorint, 2021-12-14:
I retried with this package and it worked!
https://aur.archlinux.org/packages/liquid-dsp-git/
It seems the version v1.3.2.r445.ge77c20bb, the version indicated in the last comments... from github it didn't worked but from this package yes. No I'll investigate better.

vsonnier, 2021-12-14:
Good to know, `e77c20bb` is indeed the latest version, the one I tried lately, compiling it from sratch. 
So it was a problem of pre-built binary ?

CristianRossiSorint, 2021-12-14:
Maybe it was only a version issue: the 1.3.1 (from AUR, is not 1.3.2 sorry) worked for me, but from Github I tried only 1.3.2 and 1.3.0.
I was maybe only unlucky.

vsonnier, 2021-12-14:
Liquid-dsp version numbers, do not mean much, and rarely changed, that's why only the git reference matters. 

On the other hand, using the `master` version can break previous compatibility. Cubic was adapted to the bleeding edge liquid-dsp, but it was problematic for ther users, as mentioned here : https://github.com/jgaeddert/liquid-dsp/issues/273

closed 2021-12-14 by vsonnier

jgaeddert, 2021-12-22:
Oof. Sorry. See commit ed8bb28 which adds macros for backwards compatibility.

vsonnier, 2021-12-22:
No worries @jgaeddert , as I said CubicSDR was already adapted to the new signatures :) Thnaks anyway !