cjcliffe/CubicSDR #931: Can't zoom out enough to make this useful.

bashpr0mpt, 2021-12-12:
In a dead area with a weak SDR - it takes a ridiculous amount of faffing about to find a signal, then a bunch more to the next. Allowing us to zoom out further and further would be SO much better, even if it slows the waterfall down significantly it'd be worth it. Also ever think about integrating ADS-B into this? There's so many stupid broken little things one needs to do to get the most out of their SDR, it'd be nice to have one package that does it all! There's a lot of code available for it too, it'd be a matter of the GUI fitting it more than anything.

Dantali0n, 2021-12-12:
The amount of zoom out (bandwidth) is fundamentally limited by the sampling rate (MSPS) of your SDR, nothing we can do our change except you increasing the sampling rate or getting another SDR. Technically we could show a larger bandwidth beyond the MSPS but the signals would become attenuated and prone to aliasing so there is a strong reason to not support that.

More details here:
https://www.sdrplay.com/community/viewtopic.php?t=4520

vsonnier, 2021-12-12:
What @Dantali0n said. I must add that Cubic is probably the fastest, and easiest, out there to "browse the spectrum" : use up-down arrows to zoom in, zoom out (zoom out limited by sample rate, there is no other way, this is...physics.) __and__ use left-right arrows to explore the frequency spectrum.

At 8Msps of sample rate, sweeping the 2Ghz worth of spectrum of the SDRPlay take 20s on my machine at full speed. 

You are maybe using the Frequency / Bandwidth / Center Frequency numerical controls, but you don't have to change the frequency and browse the spectrum. When using left-right arrows to sweep the frequency spectrum, the Center Frenquency adapts automatically to accomodate the current frequency range you are exploring.

closed 2021-12-12 by vsonnier