cjcliffe/CubicSDR #932: Don't compile on make: modemcf_get_demodulator_evm issue

CristianRossiSorint, 2021-12-14:
I tried to compile CubicSDR with liquid-dsp master version (1.3.2) and after with v 1.3.0, but nothing changed, it returns this error on make:
```
bios@archinator:[~/Downloads/Software/CubicSDR/build]$> make -j6
[  1%] Linking CXX executable x64/CubicSDR
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/modules/modem/ModemDigital.cpp.o: in function `ModemDigital::updateDemodulatorLock(modemcf_s*, float)':
ModemDigital.cpp:(.text+0xde): undefined reference to `modemcf_get_demodulator_evm'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1466: x64/CubicSDR] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

It seems an error referred to src/modules/modem/ModemDigital.cpp at line 52:
```
void ModemDigital::updateDemodulatorLock(modemcf mod, float sensitivity) {
    setDemodulatorLock(modemcf_get_demodulator_evm(mod) <= sensitivity);
}
```

vsonnier, 2021-12-14:
Duplicate of #930

closed 2021-12-14 by vsonnier

CristianRossiSorint, 2021-12-14:
> Duplicate of #930

I already tried to get and install liquid-sdp from master.

vsonnier, 2021-12-14:
A reopened #930. As I said there, I have no problem when both Cubic and liquid-dsp for `master`, on Windows. It may be a Linux specific problem. Please continue the discussion in #930.