cjcliffe/CubicSDR #933: Windows Build CubicSDR Errors

MaxFun, 2021-12-18:
CMake GUI errors out with ;
Could NOT find LIQUID (missing: LIQUID_LIBRARIES LIQUID_INCLUDES) 
CMake Error at C:/Program Files/CMake/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS gl core propgrid adv base)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.22/Modules/FindwxWidgets.cmake:1025 (find_package_handle_standard_args)
  CMakeLists.txt:175 (find_package)

Configuring incomplete, errors occurred

Do I need to install something else other than wxWidgets, CMake, SoapySDR?
They all appeared to install correctly.

[CMake 3.22.1 – C-MSVCDev-CubicSDR_win64 GUI output.docx](https://github.com/cjcliffe/CubicSDR/files/7738454/CMake.3.22.1.C-MSVCDev-CubicSDR_win64.GUI.output.docx)
[CMakeOutput.log](https://github.com/cjcliffe/CubicSDR/files/7738455/CMakeOutput.log)
!

vsonnier, 2021-12-18:
You have to fill some of the missing paths youself, while following the [Build Windows](https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows) guide. This guide is pretty much step-by-step, provided you use CubicSDR `master` and the latest wxWidgets versions.

You may need in CMake-gui to use 'File > Delete cache- to start over your Cubic CMake configuration, as it can happen on Cmake or Visual Studio updates.

MaxFun, 2021-12-18:
I did download the latest CubicSDR master and the latest wxWidgets versions.
Did find liquid include file in the CubicSDR master and added path after deleted cache, but not LIQUID_LIBRARIES, search computer and no LIQUID LIBRARIES anywhere
Did find wxWidgets include and libraries(lib), added path to both after deleted cache.

Configure still fails, see attachments
Any suggestions
[CMake.3.22.1.C-MSVCDev-CubicSDR error 12-18-21.docx](https://github.com/cjcliffe/CubicSDR/files/7739408/CMake.3.22.1.C-MSVCDev-CubicSDR.error.12-18-21.docx)
[CMakeOutput 12-18-21.log](https://github.com/cjcliffe/CubicSDR/files/7739411/CMakeOutput.12-18-21.log)
?

vsonnier, 2021-12-18:
These kind of setup should work:

[image]

For both Hamlib and Liquid-dsp, all you have to do is point on the right paths/files provivided in the Cubic `external` sub-directories. 

I'll update the 'Build Windows' to had those precisions when I have time.

MaxFun, 2021-12-18:
Thanks, I will set my CMake Gui
Please scroll down on your GUI and send me a screen shot of the rest of the setup, WX on down.
Thanks

vsonnier, 2021-12-18:
The bottom part is about the multiple wxWidgets libs:

[image]

As the guide sates, setting `wxWidgets_ROOT_DIR` then `Configure` should be enough to detect all the libs. If not, add `wxWidgets_LIB_DIR` mannually and `Configure` again.

MaxFun, 2021-12-18:
New problem, "Could not find a package configuration file provided by "SoapySDR""

Tried to add SoapySDR_DIR with path C:/SDR/SoapySDR_win64/cmake but there is no cmake folder in my SoapySDR_win64
Tried by adding SoapySDR_DIR with path of  C:\MSVCDev\SoapySDR\cmake\Modules which has SoapySDRConfig.cmake in it, but this error out also. 

[Error with CubicSDR 12-18-21 1017hr.docx](https://github.com/cjcliffe/CubicSDR/files/7739647/Error.with.CubicSDR.12-18-21.1017hr.docx)

vsonnier, 2021-12-18:
I don't think you are following the Windows guide at that point: it clearly states how to compile SoapySDR __first__, and how to reference the result when compiling Cubic. The previous screenshots were only about specifying library paths and includes for liquid  / Hamlib, and were not to be copied verbatim !

I don't really have much time repeating what the guide says. Just follow it, starting by building SoapySDR properly.

MaxFun, 2021-12-18:
Still getting error
Uninstalled all apps installed, all folder for Build Windows instructions. Power down computer and restarted computer.
Installed Visual Studio Community 2022; selected C++ Desktop development
Download https://github.com/wxWidgets/wxWidgets/releases/download/v3.1.5/wxWidgets-3.1.5.zip and unzip it to  C:\MSVCDev\wxWidgets
Opened wx_vc16.sln with Visual Studio Community 2022. Visual Studio Community 2022 requested retarget of solutions in VS 2010, chose no upgrade.
Choose Release and x64 for the build configuration and "Build Solution". All compile successfully.
Installed CMake
Download ZIP or clone SoapySDR from https://github.com/pothosware/SoapySDR to C:\MSVCDev\SoapySDR
Before clicking Configure, had to delete cache to allow me to select Visual Studio 17 2022" as Specify the generator for this project and "x64" as Optional platform for generator
Set CMAKE_INSTALL_PREFIX to C:/MSVCDev/SoapySDR_win64_install and SOAPY_SDR_ROOT to C:/MSVCDev/SoapySDR_win64_install
Opened "x64 Native Tools Command Prompt for VS 2022" by right-clicking and "Run as Administrator".
C:\> cd C:\MSVCDev\
C:\MSVCDEV> cmake --build SoapySDR_win64 --config Release --target clean
Response was;
Microsoft (R) Build Engine version 17.0.0+c9eb9dd64 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

C:\MSVCDev>
Next entered;
C:\MSVCDev>cmake --build SoapySDR_win64 --config Release --target install
Response was;
Microsoft (R) Build Engine version 17.0.0+c9eb9dd64 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Checking Build System
  Building Custom Rule C:/MSVCDev/SoapySDR/lib/CMakeLists.txt
  Device.cpp
... Bunch of building ...
See attached for complete response
After completion, copied file SoapySDR.dll into the C:\MSVCDev\CubicSDR_win64\x64 directory I made, did not exist yet.

download ZIP from https://github.com/cjcliffe/CubicSDR/ to C:\MSVCDev\CubicSDR
Start new CMake GUI and deleted cache.
Choose C:\MSVCDev\CubicSDR for source, C:\MSVCDev\CubicSDR_win64 for build folder and click Configure
Choose "Visual Studio 17 2022" as Specify the generator for this project and "x64" as Optional platform for generator
Set wxWidgets_ROOT_DIR to C:\MSVCDev\wxWidgets and clicked  "Configure" again
Response was ;
Could not find a package configuration file provided by "SoapySDR"
Configuring incomplete, errors occurred!

[Install SoapySDR 12-18-21.docx](https://github.com/cjcliffe/CubicSDR/files/7740005/Install.SoapySDR.12-18-21.docx)

vsonnier, 2021-12-19:
> Response was ;
Could not find a package configuration file provided by "SoapySDR"
Configuring incomplete, errors occurred!

You were on such a good path, why did you not continue the Guide instructions, then ? Why on earth do you think the successive `Configure` are for, if not for filling step-by-step the missing paths ? :)

MaxFun, 2021-12-19:
Not sure what your trying to tell me.
If the "Guide instructions" are to continue after a "Configuring incomplete, errors occurred!", it would be nice for the "Guide instructions" to indicate an error would occur and to continue.
Normally when an error occurs, you stop and fix the error, not continue.

Proceeding on,
Set SoapySDR_DIR to C:/MSVCDev/SoapySDR_win64_install/cmake , then Clicked "Configure"
Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
Where do I find"BUILD_INSTALLER' to set it to 1. "BUNDLE_SOAPY_MODS", ........ ??
Or do you mean add variable if not in CMake GUI variable window, which I did then clicked "Configure"
Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES   wxWidgets_INCLUDE_DIRS gl core propgrid adv base)
Configuring incomplete, errors occurred!
That was the last Configure again, all is not good

MaxFun, 2021-12-19:
Question; Why not set all variable at once and then only having to click Configure once?

vsonnier, 2021-12-24:
> Question; Why not set all variable at once and then only having to click Configure once?

It would be possible to dothe same as the Build Linux instructions, yes. I merely completed the guide @cjcliffe started, adding steps and details.

I thought it was the most obvious way, but apparently not :)

See, the CMake configuration is in fact hierarchical, and the CMake-gui tries to hide that and auto-configure the best-effort way:
- It presents a top-level configuration, putting in red the variables that __have__ to fill at that step,
- The next `Configure` try to continue the configuration, exposing new variables that __have__ in turn to be filled because the previous level variables were properly set, and such variables only exist when the previous level are properly set,
- Repeat.

For the user POV, it translates in a series of `Configure` where each time the user has to fill some gaps undelined in red in the GUI, until `Configure` terminates fully.

That is the spirit : be stupid, follow the instructions blindly and everything will be okay.

The drawback: It doesn't work if you try to understand what is going on 😄

MaxFun, 2021-12-24:
Thanks, I'll be stupid and start from scratch.

vsonnier, 2022-01-15:
Issue is going stale, close it.

closed 2022-01-15 by vsonnier