cjcliffe/CubicSDR #942: version 27 Make fails on Suse Tumbleweed

rskunath, 2022-03-02:
This is on the latest source, 26a worked. 

In file included from /home/rick/Sources/CubicSDR-0.2.7/src/demod/DemodulatorInstance.h:11,
                 from /home/rick/Sources/CubicSDR-0.2.7/src/demod/DemodulatorMgr.h:10,
                 from /home/rick/Sources/CubicSDR-0.2.7/src/visual/PrimaryGLContext.h:14,
                 from /home/rick/Sources/CubicSDR-0.2.7/src/CubicSDR.h:13,
                 from /home/rick/Sources/CubicSDR-0.2.7/src/CubicSDR.cpp:17:
/home/rick/Sources/CubicSDR-0.2.7/src/modules/modem/ModemDigital.h:45:53: error: ‘modemcf’ has not been declared
   45 |     virtual void digitalStart(ModemKitDigital *kit, modemcf mod, ModemIQData *input);
      |                                                     ^~~~~~~
/home/rick/Sources/CubicSDR-0.2.7/src/modules/modem/ModemDigital.h:46:54: error: ‘modemcf’ has not been declared
   46 |     virtual void digitalFinish(ModemKitDigital *kit, modemcf mod);
      |                                                      ^~~~~~~
/home/rick/Sources/CubicSDR-0.2.7/src/modules/modem/ModemDigital.h:51:40: error: ‘modemcf’ has not been declared
   51 |     virtual void updateDemodulatorLock(modemcf mod, float sensitivity);
      |                                        ^~~~~~~
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:90: CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

vsonnier, 2022-03-03:
Duplicate of #930 : As `0.27` release notes said, `liquid-dsp` v1.4+ is required now, and has the missing signatures you see.

closed 2022-03-03 by vsonnier

rskunath, 2022-03-04:
That fixed the first error (pulled and built the latest liquid-dsp) and I also needed jibjpeg8 devel. But I'm left with this yet:

[100%] Linking CXX executable x64/CubicSDR
/usr/lib64/gcc/x86_64-suse-linux/11/../../../../x86_64-suse-linux/bin/ld: CMakeFiles/CubicSDR.dir/src/modules/modem/ModemDigital.cpp.o: in function `ModemDigital::updateDemodulatorLock(modemcf_s*, float)':
ModemDigital.cpp:(.text+0xde): undefined reference to `modemcf_get_demodulator_evm'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1466: x64/CubicSDR] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:136: all] Error 2