cjcliffe/CubicSDR #949: How to read peak db of carrier?

StevenABrown, 2022-04-28:
I need to read the peak value of a carrier in the main spectrum display, so that I can compare the gain of two antennas. I can move the crosshairs of the cursor to the peak of the carrier, but there is no display of decibel value on the screen. There are no decibel values for the horizontal lines either. All I see are minimum and maximum decibel values for the scale.

righthalfplane, 2022-06-06:
I use SdrGlut to test my antennas - here is an example -

https://www.youtube.com/watch?v=JDmxGQFwX-Y&t=131s

Dantali0n, 2022-06-07:
The peak db value reported by CubicSDR or any other SDR software is useless without calibration and tuning. These devices aren't power meters they don't have a flat sensitivity across they entire band and they have automatic gain stages. They best you can do is calibrate, with automatic gain disabled, against a known source and only then read the peak db.

CubicSDR does not support these measurements as due to the constraints above they in my opinion do not really have any value.