cjcliffe/CubicSDR #951: Fedora Linux: No SoapyDSR modules were found.

c4lliope, 2022-05-09:
Hello all,

I obeyed the wiki and installed support modules for my RTL-SDR card.
I checked my install using:

```
[grace@chesapeake build]$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g00e0312c
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib64/SoapySDR/modules0.8
Module found: /usr/local/lib64/SoapySDR/modules0.8/libLMS7Support.so   (20.10.0-67edf073)
Module found: /usr/local/lib64/SoapySDR/modules0.8/librtlsdrSupport.so (0.3.2-53ee8f4)
Available factories... lime, rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

[grace@chesapeake build]$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found Rafael Micro R820T tuner
Found device 0
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  serial = 00000001
  tuner = Rafael Micro R820T
```

So, I assume I am ready to use my RTL-SDR inside CubicSDR.
Though when I open the program, I see:

[image]

c4lliope, 2022-05-10:
More background;
I obeyed the directions under: https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux#soapyrtlsdr-similar-to-other-soapy-modules
though there is no package `librtlsdr-dev` on Fedora's package manager,
so in place of `librtlsdr-dev`, I obeyed the build directions on https://github.com/librtlsdr/librtlsdr .

Dantali0n, 2022-05-11:
Use `ldd` on the CubicSDR executable to show the linked paths of executeables. Almost 100% that CubicSDR is linked against another Soapy than the one in `/usr/local/lib64/SoapySDR` as used by SoapySDRUtil.

c4lliope, 2022-05-16:
Yes, indeed;
```
[grace@chesapeake ~]$ ldd /usr/libexec/CubicSDR/CubicSDR | ag soapy
        libSoapySDR.so.0.7 => /lib64/libSoapySDR.so.0.7 (0x00007fce93df1000)
```

Sadly, I'm unsure the approach for changing this path. I am using (and had no success on):
```
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/usr/local/lib64/libSoapySDR.so.8.1"
```

because running CubicSDR keeps displaying the `/lib64/libSoapySDR.so.0.7` grade number:
```

SDR enumerator starting.
SoapySDR init..
        API Version: v0.7.1
        ABI Version: v0.7
        Install root: /usr
        Loading modules... 
        Available factories...nullJust 'null' factory found.
```

Can you recommend a second approach so I can make this change?

Dantali0n, 2022-05-16:
Your Cubic is compiled against `libSoapySDR.so.0.7` so it will never ever look for `libSoapySDR.so.0.8`. you will have to recompile Cubic and ensure the correct version of Soapy is used while compiling. Potentially the easiest way to achieve this is to ensure `libSoapySDR.so.0.8` appears in `$PATH` first.

c4lliope, 2022-05-23:
In my `~/.bashrc`, I added:
```bash
# Compile CubicSDR
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/usr/local/lib64/libSoapySDR.so.8.1"
```

I obeyed https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux again and ran a build; success!

My graphics are lagging some - the display proceeds by one-second hops. Besides this, I'm really pleased. Easy process.

closed 2022-05-23 by c4lliope

sonicepk, 2024-01-20:
dnf install soapy-rtlsdr
installs the module for rtl-sdr