cjcliffe/CubicSDR #952: 3.5 7 14 mhz RSPdx no reception, FM Broadcast, NWS weather, perfect

bradshawlupton, 2022-05-19:
a newbie question,  I get no ham HF reception,  I want ham, and 5,10,15,20 WWV for calibration of WSPR bargain tools.

I get the FM and VHF NWS beautifully.  I am using a centerfed 20' dipole to 50 ohm vnc to Antenna A on the RSPdx.

      CUbicSDR has an amazing interface.  Certainly this is user error. (it often is). 

           tnx,  Bradshaw K1TE Buzzards Bay, MA

colbyAtCRI, 2022-05-23:
Can you get any MW AM? Be sure to disable all the rf frontend filters on the RSPdx first.

bradshawlupton, 2022-05-26:
I’ll try the filters, no 5 10 15 20 MHz wwv, yet I get all of them on a little $20 am sw receiver in the driveway after dark (Boston listening to Colorado, 2000 mi)

local 50 kw 1030 am doesn’t receive.

96.1 fm   146.97 fm    162.255 nws perfect.

but, my 20 mw 10.138 wspr beacon, at 50 feet, great.

      I want my sb301 Tnx, bradshaw

colbyAtCRI, 2022-05-26:
I have a RSPduo and RSP2pro. Both have front end filters that may be switched in and out of the rf front end. These are very effective filters that remove strong commercial broadcasts while leaving Ham and SW unscathed. 

Are you using a SoapySDRPlay3 radio driver or one that is built into Cubic?

kti-priv, 2022-05-27:
I just gave this one a try, I have a small loop (80cmx200cm) with a 1:4 balun connected to my RSP1A. On 20m I see some typical QRM and a bit of FT8 with CubicSDR. 
[image]
Trying the same thing with SDRuno under Windows I get similar results:
[image]

I had to adjust a few things for fairness though 

1. Turn off auto gain on cubic SDR and use the settings shown in the image for IFGR and RFGR.
2. Turn off FFT & WF averaging in SDRuno 
3. Turned off the notches as we don"t have AM in 500kHz to 1.6MHz in Europe anymore and they also didn't change anything at 20m

I still don't understand the power measurement provided by either tool but that is typically a larger effort to figure out. 

Given that you have a 6m Dipole you may not have the best setup for reception for the given frequencies even though that is typically not a big issue.

But yes I expected to see more in the afternoon on 20m usually I get HAM traffic however.  The weekend is typically a better time.
Using my active antenna which may be a bit closer to the AM radio you mentioned gives me a different result:
[image]
Also note that I caught a bit of SSB this time.

Ultimately you should note that there are many parameters in the system you can manipulate. Typically it is helpful to change only one when comparing systems.