cjcliffe/CubicSDR #959: Linux build instructions don't lead to a working instance.

b4ux1t3, 2022-07-19:
This might end up being a SoapySDR ticket, but I have evidence that it might be on the CubicSDR side. In either case, the docs might need updating.

I was able to go through the instructions, and though I was able to complete every step, I ended up having to delve into the SoapySDR library search path to get it to even locate the module that I built (wanton softlinking was involved). . .But I did. The RTLSDR module was installed and everything came up "green". 

The resulting `SoapySDRUtils --info`:

```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-ARCH
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr
Search path:  /usr/lib/SoapySDR/modules0.8
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so (0.3.3-40d0708)
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

```

Now, when I launch CubicSDR, everything launches just fine, except I get the following in the command line:

```
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so)
  rtlsdr failed ABI check: Library ABI=0.8, Module ABI=0.8-2
        Available factories...nullJust 'null' factory found.

```

. . .and the GUI tells me that I need to install a SoapySDR module.

So, that's my evidence that the docs need updating, but now for the bit that makes me think this may still be an issue with _CubicSDR_ and not SoapySDR:

gqrx still launches just fine. 

There's no issue using whichever version of Soapy was installed when I installed gqrx (which I did from the official Arch repository, in case that is relevant). The strange thing, though, is that that error shows up in the command line output when I start gqrx now. I don't know if it did before or not, but it still works just fine.

Anyway, that's my novel of an issue report. Let me know if you need any info, and especially let me know if I need to go open an issue with Soapy regarding this. I'm rather new to SDR software, so I'm still unclear on how everything works together, and I could use a bit of help drafting the issue for Soapy.

kx1t, 2022-12-02:
@b4ux1t3 did you ever resolve this? I'm running into a similar problem trying to add the SoapySDRPlay3 driver to Soapy

b4ux1t3, 2022-12-03:
> @b4ux1t3 did you ever resolve this? I'm running into a similar problem trying to add the SoapySDRPlay3 driver to Soapy

Unfortunately, no. I haven't looked at it in a little while, though I do have plans soon-ish to try from scratch and document the changes at every step. I'll be sure to post that here when I do that.

electromage, 2023-02-01:
I'm unable to get this working either.

`$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.1-g9c4fa324
API Version: v0.8.200
ABI Version: v0.8-2
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8-2 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
`
It seems that SoapySDR is installed,  I don't know if Search path missing is related but when I try to start CubicSDR it says "No SoapySDR modules were found."

This manual compiling and installation feels very 20 years ago and not consistent with the modern Linux experience, I'm curious if this is even being maintained.

b4ux1t3, 2023-02-02:
@electromage Such is the state of a lot of ham radio projects, unfortunately. Not that they're unmaintained, but that they require a lot of hand-holding to get built. 

I'm planning on circling back around to this issue some time in the near future, seeing if I can get it resolved myself on a fresh system, try to track down where things are going wrong.

electromage, 2023-02-02:
I ended up using SDR++ with my SDRPlay and it was very fast to get going, and easy to use.

b4ux1t3, 2023-02-04:
Yeah, I ended up using gqrx on Linux for now, but I still use cubic on windows, and always regret not having it when I'm in arch.

I'm looking at this problem this week (live on stream, so that'll be fun), so I'll report back on how that goes.

simona70, 2023-02-27:
I bought a sddrplay rspdx and I have problem to find software on my Arch Linux. For now I can use only sdrpp and sdrangel.  was anyone able to use any other software?

electromage, 2023-02-27:
No, but I really like SDR++. I'll just stick with that.

cjcliffe, 2026-11-15:
building now uses build.py python script; old build instructions have been removed and docs/ contains the build guide

closed 2026-11-15 by cjcliffe