cjcliffe/CubicSDR #961: Win10 no SoapySDR modules were found

mroijer, 2022-07-26:
The error is similar to the topic "https://github.com/cjcliffe/CubicSDR/issues/411". However, I installed BUILD_INSTALLER BUNDLE_SOAPY_MODS BUNDLED_MODS_ONLY in the CMake according to the instructions. However, an error occurs. I'm trying to build CubicSDR-0.2.5 with SoapySDR-0.8.1 and CMake v3.22.1.
In the "Build CubicSDR" block, I'm using libhamlib-2.dll instead of libhamlib-4.dll in HAMLIB_DLLS and HAMLIB_LIBRARY, because version 0.2.5 does not have libhamlib-4.
Next, I making the "Using existing Modules" block, however, when launched, the application crashes without errors when trying to find the device. So I return "libliquid.dll" and "SoapySDR.dll" generated by CMake and then an endless 'No SoapySDR module found' error appears.

mroijer, 2022-09-22:
The question is still relevant. I'll pay whoever solves the problem 50$

vsonnier, 2023-01-13:
There is no point at using and outdated `0.25` version, and we cannot offer any kind of support for this.

Luckily, @cjcliffe has created automated scripts to build Cubic and dependencies [here.](https://github.com/cjcliffe/CubicSDR-WinBuild). 
Even better, there is also a pre-built [Win64 installer.](https://github.com/cjcliffe/CubicSDR-WinBuild/releases/tag/0.2.7-test2) corresponding to `0.27` release.

closed 2023-01-13 by vsonnier