cjcliffe/CubicSDR #964: Hi, does the official websit still running?

TurnOffNOD, 2022-09-04:
Hi, when I visited the official website cubicsdr.com today, the broswer says "The connection has timed out".
I am wondering is this due to my network issue, or the website is down?

SDRplay, 2022-09-04:
Working here at the moment - might have been an ISP or hosting issue?

TurnOffNOD, 2022-09-05:
> Working here at the moment - might have been an ISP or hosting issue?

Oh, now it is working. It is a temporary problem.
By the way, there is no latest release version on the website. Will the latest version of windows binary and other thing be put on the website?

cjcliffe, 2023-03-28:
Will post an update on the site when all the 0.2.7 binaries are available.

There's technically an updated windows install at https://github.com/cjcliffe/CubicSDR-WinBuild/releases but it has some liquid-dsp build related performance issues.   

That repository will hopefully contain all the automated builds in the future.

closed 2023-03-29 by cjcliffe