cjcliffe/CubicSDR #969: Can't compile CubicSDR on Mac M1 Pro,  last step

catmater, 2022-10-03:
When compile last step CubicSDR , got error: 
[  99%] Linking CXX executable x64/CubicSDR.app/Contents/MacOS/CubicSDR
ld: library not found for -lpng
clang: error: linker command failed with exit code 1 (use -v to see invocation)

But I have installed libpng (brew install libpng) already. 
Please, help, what's wrong?

supertorqued, 2022-10-07:
I have the same issue. I'm trying to build it for my Mac to address the resolution issue.

BG5VGK, 2022-11-27:
I have build the arm version, link is here:
https://github.com/BG5VGK/CubicSDR/releases/tag/0.2.7

supertorqued, 2022-11-28:
I get the dreaded "No SoapySDR modules were found." dialog when loading this version of cubic sdr. 
I pulled the head version of SoapySDR and SoapySDRPlay.  
SoapySDR built with out any issues. However SoapySDRPlay bombs when calling sudo make install with: 
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

 my libSoapySDR files are:
ls -l /usr/local/lib/libSoapySDR*
lrwxr-xr-x  1 root  wheel      23 Nov 27 20:34 /usr/local/lib/libSoapySDR.0.8-2.dylib -> libSoapySDR.0.8.1.dylib
-rwxr-xr-x  1 root  wheel  366580 Nov 27 21:49 /usr/local/lib/libSoapySDR.0.8.1.dylib
lrwxr-xr-x  1 root  wheel      23 Nov 27 20:34 /usr/local/lib/libSoapySDR.dylib -> libSoapySDR.0.8-2.dylib

Any help would be much appreciated.

BG5VGK, 2022-11-28:
I have got the same error. I tried my best to reload SDRPLAY API, but I failed. I thought it might be SDRPLAY's problem.