cjcliffe/CubicSDR #970: modemcf not declared

emrecio, 2022-10-08:
Latest version cloned from git seems to have references to modemcf that are missing all over the code base. Not sure where to start debugging? Is it a missing library?

emrecio, 2022-10-08:
``In file included from /home/erecio/elDrive/RTL/CubicSDR/src/demod/DemodulatorInstance.h:11:0,
from /home/erecio/elDrive/RTL/CubicSDR/src/BookmarkMgr.h:12,
from /home/erecio/elDrive/RTL/CubicSDR/src/forms/Bookmark/BookmarkView.h:10,
from /home/erecio/elDrive/RTL/CubicSDR/src/forms/Bookmark/BookmarkView.cpp:9:
/home/erecio/elDrive/RTL/CubicSDR/src/modules/modem/ModemDigital.h:45:53: error: ‘modemcf’ has not been declared
virtual void digitalStart(ModemKitDigital *kit, modemcf mod, ModemIQData *input);
^~~~~~~
/home/erecio/elDrive/RTL/CubicSDR/src/modules/modem/ModemDigital.h:46:54: error: ‘modemcf’ has not been declared
virtual void digitalFinish(ModemKitDigital *kit, modemcf mod);
^~~~~~~
/home/erecio/elDrive/RTL/CubicSDR/src/modules/modem/ModemDigital.h:51:40: error: ‘modemcf’ has not been declared
virtual void updateDemodulatorLock(modemcf mod, float sensitivity);
^~~~~~~
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1084: CMakeFiles/CubicSDR.dir/src/forms/Bookmark/BookmarkView.cpp.o] Error 1``

emrecio, 2022-10-08:
[image]

emrecio, 2022-12-26:
Downloaded and compiled
git clone git://github.com/jgaeddert/liquid-dsp.git
./bootstrap.sh
./configure
make
sudo make install

Worked.... IDK why I missed this before.

closed 2022-12-26 by emrecio

dmaltsiniotis, 2022-12-26:
Glad you figured it out!