cjcliffe/CubicSDR #972: Has CubicSDR been abandoned?

KenCoble, 2022-10-18:
inquiring minds want to know?

vasagle-gleblu, 2022-11-15:
That would be a shame if this project got abandoned.
The last update I can see (from the author?) is 9 months ago from the writing of this entry.

cjcliffe, 2023-03-28:
Not abandoned; unfortunately just haven't had time between work and life to get back to making updates.   Moved out to the country last October and just got to unpacking my antennas and SDRs the other day.

Going to make an effort to get automated builds going in the next couple months and start to get some issues resolved.   

I've also been working with AI tools to accelerate my workflow elsewhere, so I'm hoping to see how that can help out with this project as well.

closed 2023-03-29 by cjcliffe