cjcliffe/CubicSDR #976: Problems with liquid-dsp Version 1.5.0

srs4511351, 2022-11-22:
With with liquid-dsp Version 1.5.0 Nov 20, 2022, the Modem Spectrum and Waterfall is blank with bandwidths less than 58,890 Hz.
Additionally, when the Modem Spectrum and Waterfall display blanks, it stops reception for a second or so.
Reception seems to be normal.

I uninstalled  Version 1.5.0 and did a new install after issuing the command:
git checkout v1.4.0
It said:
HEAD is now at 3b8a6f07 Merge branch 'v1.4.0-dev'

Modem Spectrum and Waterfall display works with narrower bandwidths now.

----Steve

vsonnier, 2023-01-13:
Hello @srs4511351, can you try again with v1.50 with this Cubic commit https://github.com/cjcliffe/CubicSDR/commit/0e3a785bd2af56d18ff06b56579197b3e89b34ab ?

I encountered a very similar problem, and may have have fixed it with this commit.

srs4511351, 2023-01-13:
It is working with liquid-dsp Version 1.5.0 now, thanks!

vsonnier, 2023-01-13:
Great ! Have fun with CubicSDR.

closed 2023-01-13 by vsonnier