cjcliffe/CubicSDR #979: Latest RPi4B image crashes after install of CubicSDR

w0dz, 2022-12-11:
Latest RPi4B image crashes after install of CubicSDR

Using this image: 

2022-09-22-raspios-bullseye-armhf.img.xz

OS wakes up fine after above install. Install order is
SDRPlay API (rebooted after)
SoapySDR (tested connection to RSP1A, works)
LiquidDSP
WxWidgets 3.1.5
CubicSDR

Whole installation takes about 2  hours. On completion, script reports that CubicSDR will show up in the "Other" portion of the drop-down. On reboot, Linux crashes with a lot of blather that I can't paste here. It just had a lot of problems.

Brian

w0dz, 2022-12-12:
Is there any way to install CubicSDR on a Raspberry Pi 4B without going through all these compile steps?

srs4511351, 2022-12-13:
I have an SDR radio image for Raspberry Pi.
There are instructions to build a system and there are Raspberry Pi 4 images pre-built.
https://www.radiosrs.net/tech.html

Let me know it you get it working.

srs4511351, 2022-12-13:
I was unaware that there was a newer version of Raspberry Pi4B. What is the difference?
Mine is a Raspberry Pi 4B Rev. 1.4 and the procedure works on it. I often find that when people have trouble with my procedure, they did something different than what is specified in the procedure.
I don't actually have a script, but a procedure. It works with my Raspberry Pi 4B Rev. 1.4. The built OS image also works on a Raspberry Pi 3, which is underpowered for some of the applications.
I have built CubicSDR many times, so the problem is deeper than the scripts and procedures. Details about what went on are necessary.
raspios_arm64-2022-09-07 has a package for an older CubicSDR version 0.2.5. You can try that.
There is a problem with liquid-dsp Version 1.5.0 that I posted here. https://github.com/cjcliffe/CubicSDR/issues/976
Select an older version with git checkout v1.4.0

Does the build complete successfully? Did you install all of the many prerequisites? Which version of wxWidgets?
What are the symptoms other than just the image crashes?

I don't like the filters in CubicSDR. Signal rejection from the sideband opposite from the one you are using is poor.

w0dz, 2022-12-13:
Yeah, I found out the hard way. They make changes to the bootloader and who knows what else. It is hard to tell. All I can tell you is that the “Buster – Debian 10” images provided by SDRPlay (v0.6, v0.7) do not work – they generate this screen:

 

They work on my Pi 4B from 2 years ago but not the new ones I just received. So I installed the latest OS from RaspberryPi.org (“Bullseye” – Debian 11) and it runs fine. 

 

I then used the install scripts provided by SDRPlay (SDRplay_RPi_Scripts_v0.3) to install:

 

*	SDRPlay API (SDRplay_RSP_API-ARM32-3.07.2.run)
*	SoapySDR

Tested the connection to the RSP1A at this point, and it worked.

 

If I install pyaudio and scipy here, my python3 code works – autostart and all.

 

I also have to add the “udev” rule workaround to get headphone audio to work when you have two HDMI displays. Sheesh.

 

But then I add:

*	LiquidDSP (echo "SDRplay script to download and build liquid-dsp - v0.1")
*	WxWidgets 3.1.5 (did not try the earlier version)
*	CubicSDR (echo "SDRplay script to download and build CubicSDR - v0.3")

 

I don’t need the other stuff they make available (TCP Server, SoapyRemote)

 

It all completes after about 2 hours (grrr). But it breaks my code, and when I reboot, the OS crashes during boot. I don’t have a screen shot, sorry. I am going to try again.

 

I also tried the procedure on your web site with the same results.

 

Unrelated:

I’m also having one heckuva time making a zipped image that I can send people that can be used with the imager utility to create an SD card. I can copy the card to another one just fine (using the Pi’s Copy Disk utility in the main menu drop-down) and it boots. But when I use “sudo dd if=/dev/sda | xz > sdcard.img.xz” (where /dev/sda is the name of the copied 32G SD card) from a command prompt, it completes, but it creates a huge file (5.2G), and it takes forever to program an SD card using the imager and then doesn’t verify! The zipped image should only be around 1G. I didn’t have any better luck with gzip. If you have any ideas, I am all ears.

 

Brian

srs4511351, 2022-12-13:
What version of CubicSDR are you trying to build? CubicSDR - v0.3 does not exist and 0.2.7 is the latest.

Are you using a 64-bit OS or a 32-bit? SDRplay_RSP_API-ARM32-3.07.2.run is not appropriate for a 64 bit OS. It could crash your system.

CubicSDR was recently updated, probably v 0.2.6a to allow it to work with WxWidgets 3.1.5. Older versions won;t work with it.
I cannot use the SDRplay v0.7 image either. I did get v0.6 to work, but there are so many things it doesn't have and they are so out of date so I build my own. The picture of the boot crash did not make it into your post.
Saying that my procedure doesn't work doesn't help because it does work. I need to know what doesn't work.

I feel confident that my procedure will work on your Raspberry Pi since most of the code is built on your system, but you have to follow all of the instructions. I don't see anything that can cause it not to boot.
Again, what version of RPI do you have? It shows when you type sudo raspi-config. I can still use the latest OS.

I make a backup of my SD card on my Windows laptop and use winzip to make an archive. There are options in the imaging program that doesn't save the unused part of the disk. I have a procedure I can share.

Try this before you install soapysdr.
sudo apt install cubicsdr
It will also soapy, etc. but it's old.

w0dz, 2022-12-14:
I’m using the 32-bit OS.

 

The boot crash just said “start4.elf is not compatible. This board requires newer software.”

 

I’ve tried this so many times I am turning blue. But I will try again. Each time it crashes I learn something new. Bear in mind I have two objectives: 1) get a working OS that runs both my own python3 code using Soapy and SDRPlay’s driver, and 2) Install and run CubicSDR. The old v0.6 on the old board worked fine. But getting the same thing working on the latest board is giving me fits. I now have my own code working but had to temporarily give up on CubicSDR because I need to ship something that works to my customers.

 

I have had to take a detour to try to get a disk image of the OS+my stuff, but nothing has worked and I have gotten very good at damaging the OS using “dd” even though the input file is clearly an attached SD card and the output file is on a USB flash drive.

 

I don’t know what version of CubicSDR I am trying to get. Version 0.3 is the script version from SDRPlay. I think it should get the latest. Here’s the script:

 

#!/bin/sh

# v0.1 - first version

# v0.2 - fix read command

# v0.3 - remove read command

# Script to download, build and install the Widgets for CubicSDR

 

# clear the terminal

clear

# say what we are doing

echo "SDRplay script to download and build CubicSDR - v0.3"

# make the ~/Dev folder (but only if it doesn't already exist)

mkdir -p ~/Dev

# change to the Dev folder

cd ~/Dev

# download and install hamlib and alsa sound libraries

sudo apt install libhamlib-dev libasound2-dev -y

# remove CubicSDR folder (if it previously exists)

rm -rf CubicSDR

# download CubicSDR code

git clone https://github.com/cjcliffe/CubicSDR

# change to the CubicSDR folder

cd CubicSDR

# create the build folder

mkdir build

# change to the build folder

cd build

# create the build files

cmake .. -DCMAKE_BUILD_TYPE=Release -DUSE_HAMLIB=1 -DOTHER_LIBRARIES="-latomic" -DUSE_AUDIO_PULSE=0 -DUSE_AUDIO_OSS=0 -DUSE_AUDIO_ALSA=1 -DwxWidgets_CONFIG_EXECUTABLE=~/Dev/wxWidgets-staticlib/bin/wx-config

# start the compiler

make

# install the CubicSDR files

sudo make install

# ask the user if they want to reboot or not

echo "Finished. Visit https://github.com/cjcliffe/CubicSDR/wiki for more info"

echo " "

srs4511351, 2022-12-15:
Good move making a backup image. Make one every time you make progress. Restore it if you broke it.
None of this stuff has anything to do with the boot files. Unless you are doing something that you aren't talking about, the trouble lies elsewhere. Clearly something is changing these protected files.
Are you logging is as the root user?
From:
https://www.riscosopen.org/wiki/documentation/show/Software%20information:%20Raspberry%20Pi:%20Firmware
/boot/start4.elf – The Pi 4 GPU firmware. Forms a matched pair with fixup4.dat
/boot/fixup4.dat – Additional code for the Pi 4 GPU. Forms a matched pair with start4.elf

If you copy working copies to a USB drive and can mount your SD card in a USB programmer, you can boot on a working system and try to copy them back to see if it fixes them. Maybe copy the whole /boot directory.
In your case, try copying /boot from another working SD card to the non bootable one mounted in a USB programmer.
The most likely thing to change these files is an update, but it should get the right ones.
Does 
sudo apt update
sudo apt upgrade
cause this boot problem?

v0.3 is the script version.
I see a BIG problem with the scripts in that they compile with "make". "make -j4" will compile many times faster, but make sure you have the heat sinks installed!
Use 4buildWidgets_3.1.5.sh to install Widgets instead of 4buildWidgets.sh. You are getting the latest version of CubicSDR, which needs 3.1.5.
3buildLiquidDSP.sh installs the latest version 1.5.0 of liquid-dsp. CubicSDR does not work properly with liquid-dsp Version 1.5.0
They don't install RtAudio. CubicSDR uses this. It is listed on the CubicSDR github page.
Try CubicSDR before you reboot.
Still nothing to do with boot files.

It looks like you did not try the procedure on my web page. I follow it when I build a new system.
See: https://www.radiosrs.net/tech.html
The system installation starts under **Raspberry Pi OS Bullseye 32-bit installation...**
This installs the operating system, API, Soapy, etc. You can skip the HackRF and RTL-SDR sections.
[Installing drivers for the SDRplay RSP1A, HackRF One and RTL-SDR on a Raspberry Pi with Raspberry Pi OS](https://www.radiosrs.net/installing_drivers_sdrplay_and_HackRF_RPI_OS_armhf_bullseye.html)
Then install hamlib from [Installing utility software on a Raspberry Pi](https://www.radiosrs.net/installing_utility_software.html)
Then go to [Installing SDR applications on Raspberry Pi OS](https://www.radiosrs.net/installing_sdr_software.html).
It uses make -j4 to greatly speed up the compile.

Or just download my  32-bit Raspberry Pi OS Debian Bullseye disk image. However, I'm worried it won't work on your RPI. I would really like to know, though.
I tried one of my images on a RPI 3 and it booted!

w0dz, 2022-12-17:
Got it working by using your latest instructions. I will let SDRplay know. They need to update their website.Thanks for the help!BrianSent via the Samsung Galaxy Note10+, an AT&T 5G Evolution capable smartphone

  CubicSDR (Issue #979) 

https://www.riscosopen.org/wiki/documentation/show/Software%20information:%20Raspberry%20Pi:%20Firmware

srs4511351, 2022-12-17:
Thanks! I'm relieved for you!
Have a look around on my web site. There are a lot of cool applications and utilities.
If I can't receive or decode it, I want to know how.

w0dz, 2022-12-17:
Yes, big relief. Check out the final product at www.dzkit.com <http://www.dzkit.com>  and www.valleyhamshack.com <http://www.valleyhamshack.com> : A fully functional retro “Heathkit GR-54” lookalike, with multiple “dial pointers”, 200kHz-108MHz coverage with just a few gaps, a functional tuning knob, RFG and Volume control, and a built-in keyer. And of course, it also runs CubicSDR for those who want a traditional waterfall display.

 

Brian

w0dz, 2022-12-19:
Could I pick your brain for a minute? I have had no luck figuring out how to set the AGC parameters for the RSP1A. The documentation for the Python bindings for SoapySDR are just awful.

 

I am seeing AGC action (in my program) even when AGC is turned off. And when I turn AGC on, I get no data. Even when listening to AM, the gain comes up between words. I also can’t tell what string the api wants to control the RFGR instead of the IFGR. The gain reduction tables are a total mystery. And I can’t find any way to set the attack and decay times.

 

The Pothos doc says that the Python bindings are identical to the C API, which is clearly not true. Is there a document anywhere that shows actual an Python API, including parameters specific to various SDR’s (in my case the RSP1A)?

 

How do you do it in CubicSDR?

 

Brian