cjcliffe/CubicSDR #980: Waterfall display missing in Windows 11 install

AubreyPage, 2022-12-21:
I installed CubicSDR V 0.2.4 on Windows 10 64 bit from Github. I am using a RTL SDR dongle that appears to be genuine. At first, the CubicSDR software did not receive RF. I then installed SDR sharp. Installation of this software required the installation of a USB driver for the dongle. SDR sharp worked so that meant that my dongle was OK. I then opened CubicSDR and the receiver worked but there is no waterfall display. I have uninstalled and re-installed CubicSDR twice and even installed an earlier version but the waterfall area, both locations, is all white with no waterfall display. The waterfall function works in SDR sharp. I see no other entries in github.com/cjcliffe/CubicSDR/issues that relate to my issue. Is there an installation step that I missed? Is the waterfall function switched off somewhere?

AubreyPage, 2022-12-22:
>   install (Issue #980)
> Hi @AubreyPage, can you post a screenshot of what you're seeing? When you say there is no waterfall display do you mean that it's all dark blue/back, or that area just doesn't render at all? I've been using CubicSDR (0.2.5) on Windows 10 with an RTL SDR (and other SDRs) and have not encountered that issue before.

I will post a screenshot later today. All I see is a white background where the waterfall should be.Aubrey Sent from my Verizon, Samsung Galaxy smartphone

AubreyPage, 2022-12-22:
Attached find a screenshot of the CubicSDR app open. I hear audio just fine and the RF receiver functions seem normal.
[image]

AubreyPage, 2022-12-22:
I did not intend to close this. I intenede to close only. I don't know how to correct my error. Sorry.

bVidDa, 2022-12-22:
I have the same problem since I updated to Windows 10 22H2, it was previously working with 21H1. I also updated the Graphic Drivers the same day (AMD 5700), so I'm not sure which of the two updates to blame. I bet for Windows 10. It's been a while, and I still haven't been able to fix it. It happens with all versions (0.2.4, 0.2.5), including 0.2.6a from phothosware. Logs don't show any warnings about this. My screenshots would be very similar, a White Area where there should be Waterfalls.

srs4511351, 2022-12-27:
The latest version of liquid-dsp (v1.5.0) causes a problem just like that on Raspberry Pi.
What version of liquid-dsp do you have? 'v1.4.0 works for me.
I haven't installed CubicSDR on Windows recently, so I don't know how liquid-dsp is installed on Windows
See issue: Problems with liquid-dsp Version 1.5.0 #976.

AubreyPage, 2022-12-28:
How would I find the version of liquid-dsp that got installed on my PC during the CubicSDR installation? I looked in /program files/CubicSDR and did not see any file with liquid-dsp in the name. Should liquid-dsp be installed separately? Maybe this is the issue?

srs4511351, 2022-12-28:
I assume that you used the CubicSDR-0.2.4-win64.exe installer. CubicSDR V 0.2.4 is an old version from 2018.
It would install libliquid.dll for you, it should be the right one for that version of CubicSDR. Unfortunately it does not contain the version.
The good news is that it is too old to be version 1.5.0.
libliquid.dll and the other files are in C:\Program Files\CubicSDR
Of course, your SDR drivers will have to be installed first.
I found newer versions:
https://github.com/cjcliffe/CubicSDR/releases/download/0.2.5/CubicSDR-0.2.5-win64.exe
https://github.com/cjcliffe/CubicSDR/files/3346714/CubicSDR-0.2.6.a1-win64.exe.zip

I installed 0.2.6.a1 on Windows 10 64 bit 21H2 and it works.
Check that you have C:\Program Files\CubicSDR\libliquid.dll

srs4511351, 2022-12-28:
The Pothos SDR environment installs CubicSDR v0.2.6a and other things that may be useful. It has everything you need to run their program selection.
https://github.com/pothosware/PothosSDR/wiki/Tutorial

The download page:
https://downloads.myriadrf.org/builds/PothosSDR/

AubreyPage, 2022-12-28:
Reply to srs4511351: I installed from github using CubicSDR-0.2.4-win64.exe. In your above post you say that I should install the drivers first. I did not know to do that. CubicSDR would nit receive RF at first. I then installed SDRsharp and in doing so, it installed USB drivers and perhaps other drivers. When I went back to tru CubicSDR after that, it worked in the sense taht it receives radio but there is no waterfall. I checked the program files/CubicSDR  directory and the libliquid.dll file is present. I am running Windows 11 home 64 bit V22H2 and AMD Ryzen 5 3600 processor and ASRock Phantom RX570 4G/S graphics with the latest drivers installed. 

Do I need to remove CubicSDR and install new drivers and then re-insta;; version 0.2.6.a or maybe 0.2.5? When I went to Github, v 0.2.5 was available only in source code, not a win 64 installer.

b-crutch, 2022-12-28:
I am experiencing the same behavior where the waterfalls are empty white regions.  Running Win 11 Enterprise v21H2.

[image]

AubreyPage, 2022-12-28:
Thanks, that helps. It shows that v0.2.6a does not resolve the issue and that it seems to be systemic issue on windows 11.

srs4511351, 2022-12-29:
The CubicSDR github site lists the software that it uses, but apparently the device driver is too basic...
The newer CubicSDR installers that I included in my post are available at the links I provided. Version 0.2.7 is available in source code.
There must be a conflict with openly or fftw3 with Windows 11. It does work in Windows10.
You probably don't need to reinstall the drivers, but uninstall and reinstall CubicSDR.
You can try the Pothos suite as it includes gqrx, which is one on the best.

srs4511351, 2022-12-29:
Based on the comments in this thread and what I found on the Internet, I put together a summary.
I found a post on Reddit that says they have CubicSDR working on Windows 11, but did not mention a version.
The only two versions of Windows 11 are 21H2 (original release) and 22H2 (2022 Update)
@b-crutch can you double-check your Windows version?
There is a discrepancy as to whether or not Windows 11 works with CubicSDR.
It looks like Windows 11 is NOT the problem, but possibly the 22H2 update in either Windows 10 or 11.
I am updating my Windows 10 to 22H2. I expect that my CubicSDR waterfall will not be working.

**Works:**
Windows 10 21H1
Windows 10 64 bit 21H2
Windows 11, unknown version

**Does not work:**
Windows 10 22H2
Windows 11 Enterprise v21H2 - please check version
Windows 11 home 64 bit V22H2

b-crutch, 2022-12-29:
I am running Win 11 Enterprise 21H2.  Winver.exe shows:

Microsoft Windows
Version 21H2 (OS Build 22000.1335)

@ Microsoft Corporation. All rights reserved.
The Windows 11 Enterprise operating system and its user interface are
protected by trademark and other pending or existing intellectual property
rights in the United States and other countries/regions.

srs4511351, 2022-12-29:
That's odd. A user on reddit claims to have CubicSDR working on Windows 11 and that would most likely be 21H2, as the only other version is 22H2. That version fails on both Windows 10 and 11.
See: https://www.reddit.com/r/RTLSDR/comments/zj0mos/windows_11_fail/
There isn't much out there about this issue.

b-crutch, 2022-12-29:
I just got this Nooelec SMART dongle a few days ago for Christmas.  I had never used an SDR prior to this, and attempted to use the application they link to on their page (which is CubicSDR).  Everything seems to work when I run CubicSDR EXCEPT for the waterfall.  I can click around within the white "waterfall" region and hear various channels / frequencies ... just no actual waterfall visualization.  After trying to troubleshoot (unsuccessfully), I downloaded SDR++ which has worked fine.  Cubic looks pretty nifty, so I'll circle back if/when a resolution is discovered.  ¯\_(ツ)_/¯

srs4511351, 2022-12-29:
After my upgrade to Windows 10 Enterprise 22H2, My CubicSDR still works!
One difference may be that I am using soapy remote to play my SDRs connected to me Raspberry Pi.

I wonder why @bVidDa installation broke when they updated to Windows 10 22H2? Maybe it was the Video driver?
Could you roll back the video driver to see if that fixes it?

TurnOffNOD, 2022-12-29:
@AubreyPage 
@bVidDa 
@srs4511351 
@b-crutch 

Hi, there is a newer version of windows installer, the repository is: https://github.com/cjcliffe/CubicSDR-WinBuild/releases . The latest version number is 0.2.7-test2-hamlib at the time I post.

Have you guys tried this version on Win 11?

AubreyPage, 2022-12-29:
I just used RevoUninstaller and un-installed CubicSDR and all of it's root folders and registry keys. I then downloaded v0.2.7 from
https://github.com/cjcliffe/CubicSDR-
[image]
WinBuild/releases. 

The waterfall still does not work. Reminder, I am running Windows 11 home 64 bit V22H2 and AMD Ryzen 5 3600 processor and ASRock Phantom RX570 4G/S graphics with the latest drivers installed. So a reply to TurnOffNod, above, yes I tried v0.2.7 on Windoiws 11, v22H2. Screenshot included.

b-crutch, 2022-12-29:
v0.2.7 does not resolve the missing waterfall on my computer, either.  My experience remains identical to that reported by @AubreyPage.

srs4511351, 2022-12-30:
This seems to be a graphics related issue, so I looked at changes related to that in the Windows 22H2 upgrade.

Windows 11 22H2 includes the new version of Microsoft’s GPU architecture “WDDM” (Windows Display Driver Model). Version WDDM 3.1 comes with performance improvements and a couple of unknown changes that will be revealed later.
It should be backward compatible with older drivers that do not support this.
WDDM 3.1 made it necessary for the graphics hardware manufacturers to release new drivers for Windows 11 22H2. They can be installed before the upgrade to 22H2.

It would be interesting to know if people who are having trouble with the CubicSDR waterfall have these new graphics drivers.
If someone wants to roll back their video driver to a version from well before the Windows 22H2 upgrade, it may help.
My graphics drivers are too old and there are no updates, so  I cannot try this.

I upgraded to Windows 10 64 bit 22H2 and CubicSDR still works. CubicSDR stopped working with the waterfall for other others when they upgraded to Windows 10 64 bit 22H2.

AubreyPage, 2022-12-30:
I am running Windows 11 v22H2. I installed CubicSDR and started this thread when the waterfall display didn't work. Later, I installed the latest AMD graphics drivers v22.11.2. I do not know the previous version but it was old. The waterfall display still does not work. This morning, I verified that I have the latest AMD/ASRock drivers. Windows update says I also have the latest version of Windows 11, no updates are available. So I have verified no waterfall before and after graphics driver updates on Windows 11 v22H2.

bVidDa, 2023-01-04:
Unlike @AubreyPage, I can't confirm if it worked with my older drivers, but it might if they were old enough. It is possible that there is a combination of GPU and Drivers that will work with the 22H2 versions.

Sorry but I don't plan to rollback my drivers right now as I have a WHQL version, which have given me good long term results so far. I'm still hoping to find some workaround.

By the way, I am also using Soapy Remote, to access a HackRF. That doesn't seem to matter, I've also tried with an Rtl-Sdr. I would say that, based on the data we have so far, @srs4511351 is on the right track with his last comment.

I Tested every version from 0.2.4 to 0.2.7, with AMD Drivers 22.11.2 and Windows 10 22H2.
I stick to 0.2.5 because the others are not giving me good performance.

b-crutch, 2023-01-04:
I don't know if this will help, but I'll provide some additional details about my laptop.
[image]
[image]

vsonnier, 2023-01-13:
Hello all, 

Apparently the problem is with AMD only vs. OpenGL on Windows 10+.  _Sigh._ 

Alas, AMD OpenGL has been a piece of shit for 10+ years at least, to the point that I know of OSS projects that stopped supported it. The situation is better on Linux form what I've heard.

Anyway, I may have a patch for you to try for Windows10 / 11.

Could you try the following:

- Install the `0.2.7-test2-hamlib` found  here: https://github.com/cjcliffe/CubicSDR-WinBuild/releases
- Override with the following patch: [CubicSDR_v028a_patch_Win64.zip](https://github.com/cjcliffe/CubicSDR/files/10409268/CubicSDR_v028a_patch_Win64.zip) based on Master https://github.com/cjcliffe/CubicSDR/commit/e09e7e8381acc753fa35bee870e977a0f9dc4321

srs4511351, 2023-01-13:
I found this about the one of the AMD drivers.
AMD 22.7.1 driver has major OpenGL optimizations and Windows 11 22H2 support
It contains support for the upcoming major Windows 11 update 22H2, feature expansions, the long-time-coming OpenGL optimizations, and more.

vsonnier, 2023-01-13:
It doesn't neccessarily mens an improvement for us : Perhaps they are going to break the __old__ OpenGL API calls (< 3.1 but I'm not sure) used in CublicSDR even more by supporting it even less, to favor __modern__ OpenGL (4.5+) intead.

b-crutch, 2023-01-14:
I'm normally pretty competent with Windows, but I'm using a new Win11 laptop and I cannot figure out how to get that patched executable out of the zip file and into the program file folder.  Some Windows security feature is arguing with me (and I'm losing).  It's  aggravating that a 30-second file copy turns into a research project 😣.  I'll keep you posted if I figure it out and can confirm / deny this patch solves the "missing waterfall" mystery.

srs4511351, 2023-01-14:
I had to extract the 2 files to a temporary location, then move them into C:\Program Files\CubicSDR
It never did let me copy from the archive directly.
It will ask for administrative privileges.
Another way would be to open an administrator command prompt and copy them with that.
I may help to trash the old files in C:\Program Files\CubicSDR first.

bVidDa, 2023-01-23:
The patch does not solve the issues for me, confirmed.
I have problems with performance using 2.7, anyway
This may be fixed by an AMD or Windows update,
if they give this the attention it deserves.

Roturbo, 2023-06-14:
Hi have win10 22H2, same problem,,,..

vsonnier, 2023-06-14:
Try to install https://github.com/pal1000/mesa-dist-win following the instructions. This is a Windows build of Mesa, using D3D12 by default to emulate OpenGL with excellent performance. 

Maybe it will be more and more the way to go for an increasingly deprecated OpenGL in the future years.

Roturbo, 2023-06-18:
Tanks @vsonnier , i try it, chose the per-application, pointed the path and name of the application 
but still didn´t work for me, i will wait for better solution.

Regards.

stefan-mrtc, 2023-08-18:
I had the same issue with Windows 11 22H2 and AMD GFX.
Processor	AMD Ryzen 7 5825U with Radeon Graphics, 2000 Mhz, 8 Core(s), 16 Logical Processor(s)
Just downloaded the latest AMD drivers (Via AMD Adrenalin) and waterfall issue is resolved.
New GFX Driver Version	31.0.21023.2010

closed 2026-11-15 by cjcliffe