cjcliffe/CubicSDR #985: Strange tuning effect with mouse wheel

srs4511351, 2023-01-13:
Running CubicSDR v0.2.8a
Raspberry Pi on DietPi 64 bit Debian

When tuned to an FM station with the FMS demod, the station is initially displayed correctly on the spectrum/waterfall.
If I change the Center frequency 100 KHz digit with the mouse wheel and it increments by 2, the original station appears on the display 100 KHz off.
If I the Center Frequency goes up 200 KHz, the station appears 100 KHz high. Going down 200 KHz shows the station as 100 KHz low. The station is distorted as expected.

Changing the Center Frequency by 100 KHz restores it to normal operation. 100 KHZ increments work consistently.

Sometimes double-clicking on the digit to change it by 200 KHz produces a similar result.

I have not seen this on previous versions.

oregonjim, 2023-04-26:
CubicSDR v0.2.8
Raspberry Pi 4
Raspberry Pi OS "bullseye" 64-bit

When using the mouse wheel to change digits on Frequency, Bandwidth, or Center Frequency, the digits always jump by two instead of one.  This makes it impossible to tune to an exact frequency.

Note that CubicSDR 0.2.8 does not seem to have this issue under Intel/Fedora Linux 64-bit.

cjcliffe, 2026-11-14:
There was a race with the frequency command queue that's now fixed on master for 0.3.0 to make sure that center frequency actually lands where it should and doesn't leave modems shifted out of place

closed 2026-11-14 by cjcliffe

srs4511351, 2026-11-15:
I installed CubicSDR v0.3.0-alpha tonight.
The frequency still often increments by 2 using the mouse wheel. Sometimes in increments by 1.
It the fix in this release?