cjcliffe/CubicSDR #986: Building on MacOSX (Venture) - Make warning

oxoocoffee, 2023-01-13:
Hello,

I am attempting to build CubicSDR-0.2.7 using cmake version 3.25.1 . After installing brew packages
*  wxwidget
*  soapysdr
*  liquid-dsp

I did try to run came from build/ I got this warning
```
-- Found OpenGL: /Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/OpenGL.framework
CMake Warning (dev) at /opt/homebrew/Cellar/cmake/3.25.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (LIQUID)
  does not match the name of the calling package (Liquid).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindLiquid.cmake:20 (find_package_handle_standard_args)
  CMakeLists.txt:171 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

So I adjusted ```cmake/Modules/FindLiquid.cmake``` and changed it from

```
find_package_handle_standard_args ( LIQUID DEFAULT_MSG LIQUID_LIBRARIES LIQUID_INCLUDES)
```

to

```find_package_handle_standard_args (Liquid DEFAULT_MSG LIQUID_LIBRARIES LIQUID_INCLUDES)```

Got rid of this warning

Cheers

mgiugliano, 2023-01-14:
@oxoocoffee The homebrew procedure (https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX---Homebrew) worked under macOs Ventura 13.1 (22C65) on an Intel Mac (16-inch, 2019). CubicSDR crashed once, but now it seems working.

oxoocoffee, 2023-01-14:
I do not see latest version in home brew repo. That is why I attempted building from source. Who is maintaining brew package?
Also I am on M1 not in Intel.

ackerthehacker2, 2023-01-22:
M1 on 13.2 built with warnings ..but working ...used homebrew too ...

luzpaz, 2023-11-17:
macos label needed