cjcliffe/CubicSDR #989: Cubik SDR has lower sensitivity compare to SDR# on same frequencies.

Perekur, 2023-02-17:
With compare to SDRSharp - Cubik has lower sensitivity for the some range of frequencies.
I found that for example on 466Mhz I can't catch signal even if on SharpSDR I can look strong signal without amplifier option. 

The configuration is next:
HackRF One (driver v 2021
CubicSDR for Windows 10 (v0.2.4)
SharpSDR (v1.0.0.1881)

I use same antenna and same HackRF driver.
The VGA and LNA settings was set similar to both applications.

[image]

[image]

dmaltsiniotis, 2023-02-17:
Hi @Perekur ,

Just a guess here but what you are seeing might be expected behavior. In your screenshot of the CubicSDR, I noticed that you're tuned to EXACTLY the center frequency:
[image]
You might have problems doing this because with the HackRF One (and almost all other SDRs) the very center of the FFT contains what is known as a "DC Spike" which is a artifact of quadrature sampling math.

Try this. Tune your center frequency 1 MHz above or below your actual target, then point to the de-modulator (click and drag it) to that section you are interested. What you might find is that it's no longer being filtered out (the dip is a give-away that there is "DC Correction" or "DC Filtering" going on.

SDR Sharp might be doing this automatically, which would account for the difference.

Perekur, 2023-02-18:
Thank you for the hint.. but it does not change the situation.

Essentially I want notice that when I using zoom (Shift + Mouse scroll) on 446Mhz
 it also impact on signal streight, and resposability of UI decrease - quite freeze. Maybe here the root cause.

[image]

SpaceCadet1249, 2024-03-23:
Hello. As a Mac M2 (Apple Silicon) user, I'd like to chime in on this topic. I can't use SDR++ because it does not support my SDRplay receiver. So, for comparison, I'm using the SDRconnect program. Observe the signal and audio in all three example video's I'm including. 1. Is my Yaesu FTDX1200. 2. Is CubicSDR. 3. is SDRconnect.
Bear in mind that all three are tuned into the 15Mhz WWV signal within 5 minutes of each other.

https://github.com/cjcliffe/CubicSDR/assets/164538664/ab98a495-c016-4994-a8ef-d412c5f897c0

https://github.com/cjcliffe/CubicSDR/assets/164538664/63f2bc1c-b910-41d1-99c4-81d89f29fcf8

https://github.com/cjcliffe/CubicSDR/assets/164538664/81358b25-f00a-484b-9ba4-2fa7ba2c6833