cjcliffe/CubicSDR #990: Windows build

batica81, 2023-02-25:
It would be nice to have a Windows build for recent releases as well.

cjcliffe, 2023-04-18:
Some test builds are now available to try at https://github.com/cjcliffe/CubicSDR-WinBuild/releases/tag/0.2.8-test1 

Hope to have automatic builds go there and eventually promote successful ones to release page in main CubicSDR repository for each release version.

dmaltsiniotis, 2023-04-18:
Not sure if this is the best place to report issues, but I tried the latest test build linked above and it crashes for me while enumerating devices:
[image]

cjcliffe, 2023-04-18:
Reporting issues here should be fine; I won't necessarily respond to each report though -- I'm looking for overall common problems at the moment.

Haven't installed the latest build here yet myself but will test it after work.

cjcliffe, 2023-04-18:
Rebuilt 0.2.8 again; updated builds are here: https://github.com/cjcliffe/CubicSDR-WinBuild/releases/tag/0.2.8-test-builds-3

This one updates to Boost 1.82; the resulting DLLs aren't getting flagged by Bitdefender anymore.

dmaltsiniotis, 2023-04-18:
Awesome, thanks for such a quick turn-around. I tested this new installer and it went _much_ faster than the last one. Also, no issues starting up this time. The FFT has a slightly different appearance/feel to it. I'll keep poking around and testing!

cjcliffe, 2023-04-18:
I'm guessing the new MSVC distributable already being installed from the last run made things quicker at least.

Bitdefender still telling me that my newly built Boost 1.82 DLLs have malware "Gen:Variant.Tedy.342071" which appears to be a non-existent heuristic match -- full scan on the build system reveals nothing there.  

Going to try a few pre-built packages from the boost download site; if they don't get flagged I'll update WinBuild to use those instead.

cjcliffe, 2026-11-14:
Circle CI now builds Windows / Mac (arm and x64) / Linux deb and AppImage -- builds will be available for every tagged release now

closed 2026-11-14 by cjcliffe