cjcliffe/CubicSDR #993: Apple macOS binary

FleXoft, 2023-03-10:
It would also be nice to have an Apple macOS build for the recent releases as well!

cjcliffe, 2023-04-18:
CubicSDR-WinBuild appears to be working well now for automatic release builds.   

Will be working on a similar build repository for macOS soon; so hopefully everything will be back in sync when that's complete.

FleXoft, 2023-04-18:
Thnx. 🤞

dmbarry86, 2023-05-29:
Thanks for the great tool @cjcliffe . Would it be possible to have a Mac OS build for the latest 0.2.7 release?

rkarikari, 2023-11-25:
Hello 
I get the followig with the latest macos build...

[image]

but I have soapysdr installed..

[image]

My system ...

[image]

Any feedback would be greatly appreciated.

Thank you so much.

chenrui333, 2023-11-27:
👋 Any update for the macos binary for the 0.2.7 release? Thanks!

cjcliffe, 2026-11-14:
Binaries will build for all platforms for each tagged release via CircleCI now; keep an eye out for 0.3.0 bins

closed 2026-11-14 by cjcliffe