cjcliffe/CubicSDR #996: CMake breaks. Build CubicSDR not possible.

ILabe, 2023-04-07:
At the step: [Build CubicSDR](https://github.com/cjcliffe/CubicSDR/wiki/Build-Windows) it comes to abort with the error message after I set the environment variable: **wxWidgets_ROOT_DIR**

The error message in the C-Make GUI 3.25.3:
```
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
64 bit compiler detected
CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (LIQUID)
  does not match the name of the calling package (Liquid).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindLiquid.cmake:20 (find_package_handle_standard_args)
  CMakeLists.txt:171 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.25/Modules/FindwxWidgets.cmake:934 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "version" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  CMakeLists.txt:175 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at C:/Program Files/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find wxWidgets (missing: gl core propgrid adv base) (found
  version "3.2.1")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.25/Modules/FindwxWidgets.cmake:977 (find_package_handle_standard_args)
  CMakeLists.txt:175 (find_package)

Configuring incomplete, errors occurred!
See also "C:/MSVCDev/CubicSDR_win64/CMakeFiles/CMakeOutput.log".
```

What should the user do next and shouldn't we adjust the installation instructions?

cjcliffe, 2023-04-09:
Make sure that wxWidgets_ROOT_DIR is pointing to the folder you actually unzipped the wxWidgets-3.2.1.zip to

The path specified in the instructions assumes you unpacked the zip and renamed the folder to C:\MSVCDev\wxWidgets

levicki, 2023-08-17:
There is also an issue of instructions telling us to build wxWidgets as a static lib which doesn't build wxrc.exe which is then missing and CMake can't find it.