cjcliffe/CubicSDR #1008: Add support for the RTL-SDR Blog V4 to AppImage and dmg packages

rtlsdrblog, 2023-08-20:
Hi, I've just released the RTL-SDR Blog V4 (https://www.rtl-sdr.com/rtl-sdr-blog-v4-dongle-initial-release/) which is based on the R828D tuner chip. But the hardware is slightly different as it uses a different LO frequency for the R828D compared to what is hardcoded in the drivers, and the RF inputs need to be switched in a different way.

I've created a fork of the Osmocom drivers here https://github.com/rtlsdrblog/rtl-sdr-blog-minimal, and a more heavily modified one here with other unrelated changes https://github.com/rtlsdrblog/rtl-sdr-blog. These repo's were modified to add support for the V4, and the -minimal fork has the minimal amount of changed to make it work. I've submitted a patch to Osmocom to try and upstream the main changes for the V4, but the Osmocom rtl-sdr project might be dead as I see that patch requests have been unanswered for a few years. There are no changes that break compatibility with other RTL-SDR dongles.

If it's not upstreamed is there some other way to support the RTL-SDR Blog V4?

On Windows it's easy as I can just tell users to replace the .dll file, and if installed natively on Linux, users just need to update the system RTL-SDR driver. But for solutions that prepackage the rtlsdr driver together with the software like AppImage and dmg for Mac's there is a problem as the drivers are not so easy to swap out.

rtlsdrblog, 2023-08-22:
Just noting that Osmocom replied to my upstream request so it's likely that it will be upstreamed soon. So probably the only changes required will be pulling the Osmocom code when its patched, and updating the packages.

kieran, 2023-12-15:
Was this ever upstreamed by Osmocom? Would love to use Cubic with my v4 :-)