cjcliffe/CubicSDR #1009: Appimage crash while staring under Debian 11

andres-hurtado-lopez, 2023-09-25:
Hi:

After downloading the app image and enabling it as an executable the following error appear while trying to start:

```
andresh@andres-hurtado-lopez:~/appimages$ ./CubicSDR-0.2.4-x86_64.AppImage
./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libthai.so.0: version `LIBTHAI_0.1.25' not found (required by /lib/x86_64-linux-gnu/libpango-1.0.so.0)
./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib/x86_64-linux-gnu/libgio-2.0.so.0)
./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib/x86_64-linux-gnu/libmount.so.1)
```

Some general background of the computer I'm running this

```
andresh@andres-hurtado-lopez:~/appimages$ uname -a
Linux andres-hurtado-lopez 6.5.0-1-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.5.3-1 (2023-09-13) x86_64 GNU/Linux

andresh@andres-hurtado-lopez:~/appimages$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux trixie/sid
Release:	n/a
Codename:	trixie
```

I could install those dependencies by that completely goes against the whole idea of using and Appimage.

cjcliffe, 2023-10-18:
Try updated build at https://github.com/cjcliffe/CubicSDR-AppImageKit/releases/tag/0.2.8-test2 and let me know if it does any better

andres-hurtado-lopez, 2023-10-19:
Thank you so much. I can confirm that it is properly running now.  Thank you so much !!!!!

Here are some screen captures.

[https://ibb.co/TR6TjFK](https://ibb.co/TR6TjFK)
[https://ibb.co/f0cfqMB](https://ibb.co/f0cfqMB)

I also would like to share the STDOUT of my terminal as I run it , there where some warnings, but nothing that  I felt that was degraded the performance of the software: 

```
$ ./CubicSDR-0.2.8-x86_64.AppImage
Loading:: configuration file '/home/andresh/.CubicSDR/config.xml'
Gtk-Message: 22:31:23.438: Failed to load module "xapp-gtk3-module"
Loaded 287 rig models via hamlib.

Audio Device #0 USB2.0 Device Mono
	Default Output? No
	Default Input? Yes
	Input channels: 1
	Output channels: 0
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		24-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz
		192000hz

Audio Device #1 Family 17h/19h HD Audio Controller Speaker + Headphones
	Default Output? Yes
	Default Input? No
	Input channels: 0
	Output channels: 2
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		24-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz
		192000hz

Audio Device #2 Monitor of Family 17h/19h HD Audio Controller Speaker + Headphones
	Default Output? No
	Default Input? No
	Input channels: 2
	Output channels: 0
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		24-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
[... 220 lines cut ...]
	Default Output? No
	Default Input? No
	Input channels: 2
	Output channels: 0
	Duplex channels: 0
	Native formats:
		16-bit signed integer.
		24-bit signed integer.
		32-bit signed integer.
		32-bit float normalized between plus/minus 1.0.
	Supported sample rates:
		8000hz
		16000hz
		22050hz
		32000hz
		44100hz
		48000hz
		96000hz
		192000hz

** (CubicSDR:2552923): WARNING **: 22:31:48.647: atk-bridge: get_device_events_reply: unknown signature

(CubicSDR:2552923): Gtk-WARNING **: 22:33:05.408: Could not load a pixbuf from /org/gtk/libgtk/theme/Adwaita/assets/bullet-symbolic.svg.
This may indicate that pixbuf loaders or the mime database could not be found.
```

closed 2023-10-19 by andres-hurtado-lopez

cjcliffe, 2023-10-20:
@andres-hurtado-lopez thanks for confirming; glad that works -- hoping to have all the 0.2.8 builds up soon