cjcliffe/CubicSDR #1011: macOS Sonoma Crash

overbog, 2023-10-05:
Crashes at startup:

Crashed Thread:        4  Dispatch queue: CA DispatchGroup

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000000024a
Exception Codes:       0x0000000000000001, 0x000000000000024a

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [17843]

jmohin, 2023-10-06:
Same for me:

Crashed Thread:        14  Dispatch queue: CA DispatchGroup

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000000024a
Exception Codes:       0x0000000000000001, 0x000000000000024a

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [2467]

VM Region Info: 0x24a is not in any region.  Bytes before following region: 140737488227766
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      shared memory            7ffffffe1000-7ffffffe2000 [    4K] r-x/r-x SM=SHM

daBee, 2023-10-10:
Mac Mini M1, Sonoma.  

PC register does not match crashing frame (0x0 vs 0x100D2D2D0)

Crashed Thread:        35  Dispatch queue: CA DispatchGroup

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000000024a
Exception Codes:       0x0000000000000001, 0x000000000000024a

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [64585]

VM Region Info: 0x24a is not in any region.  Bytes before following region: 140722850856374
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      mapped file              7ffc97898000-7ffcbcfb8000 [599.1M] r-x/r-x SM=COW  ...t_id=120a9d93

Error Formulating Crash Report:
PC register does not match crashing frame (0x0 vs 0x100D2D2D0)

sidewinder666, 2023-10-11:
2021 Macbook Pro M1, upgraded to Sonoma and now CubicSDR crashes at startup with similar message the others have already posted:

Notes:
PC register does not match crashing frame (0x0 vs 0x100E3E2D0)

Crashed Thread:        15  Dispatch queue: CA DispatchGroup

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Codes:       0x0000000000000001, 0x0000000000000008

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [13403]

lagomorph, 2023-10-13:
same on 2019 MBP upgraded to Sonoma

dtanonis, 2023-10-15:
Same here 2020 MBP after Sonoma upgrade, reopen doesn't take

Process:               CubicSDR [1801]
Path:                  /Applications/CubicSDR.app/Contents/MacOS/CubicSDR
Identifier:            com.cubicproductions.cubicsdr
Version:               0.2.4 (0.2.4)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-10-15 10:39:53.9834 -0700
OS Version:            macOS 14.0 (23A344)
Report Version:        12
Bridge OS Version:     8.0 (21P365)
Anonymous UUID:        C4FFCAF1-2037-F15C-C919-00C898E99893

Sleep/Wake UUID:       13F80E1B-2976-4E1E-87EC-1DD6E5E92760

Time Awake Since Boot: 15000 seconds
Time Since Wake:       1548 seconds

System Integrity Protection: enabled

Crashed Thread:        3  Dispatch queue: CA DispatchGroup

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000000000024a
Exception Codes:       0x0000000000000001, 0x000000000000024a

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1801]

rhaguiuda, 2023-10-18:
Same on Intel i7

timmattison, 2023-11-01:
Tried compiling it from scratch this morning and it works again for me. You'll need all of the dependencies installed first but usually I do this for a new C/C++ project on MacOS:

```
git clone whatever...
cd whatever
mkdir build
cd build
cmake ..
```

If that doesn't report any errors then just run `make`. If it does report an error like Liquid DSP is missing I just do:

```
brew search liquid
```

And see if there's a package for it. If there is I install it and try again. And that's how I got CubicSDR running on my machine. It's awful, but it works.

cixio, 2023-11-03:
Since the current unreleased version seems to be without this bug (and I can't manage to build it myself), would it make sense to create a new release? After all, the old one is already over 1.5 years old. @cjcliffe

bobbyl140, 2023-11-20:
+1 on OP, will try building

bobbyl140, 2023-11-20:
Never mind, I can't figure out these dependencies.

bobbyl140, 2023-11-20:
For any still interested, 0.2.3 opens right away for me on Sonoma. Try that one!

sidewinder666, 2023-11-20:
And I'll confirm that, v0.2.3 works fine on Sonoma (M1 Macbook Pro).
That begs the question "What broke?" in 0.2.4 and 0.2.5.

bobbyl140, 2023-11-20:
Without looking at any of the code, my first guess would be dependency updates, but take that with a grain of salt.

cixio, 2023-11-20:
> For any still interested, 0.2.3 opens right away for me on Sonoma. Try that one!
Thanks, works also on my 2021 14'' M1 Pro

> And I'll confirm that, v0.2.3 works fine on Sonoma (M1 Macbook Pro). That begs the question "What broke?" in 0.2.4 and 0.2.5.
The actual code works also great (if you build it yourself), but I couldn't get the soapy modules running as I dont understand the build manual.

vk2tty, 2023-12-31:
This covered my attempt:

```sh
brew install cmake liquid-dsp wxwidgets
```

… but I'm not sure what I already had installed that might have helped. Anyhow, got b959ff compiled and working on macOS "Sonoma" 14.2 (23C64).

radito, 2024-03-03:
Just finished building v0.2.8 for Apple ARM, if somebody want's to try
Running some test right now on my m1 sonoma 14.3.1 and it works fine

https://github.com/radito/CubicSDR/releases/tag/0.2.8

vk2tty, 2024-03-04:
> "CubicSDR.app" can't be opened because Apple cannot check it for malicious software. This software needs to be updated. Contact the developer for more information.

[image]

radito, 2024-03-04:
> > "CubicSDR.app" can't be opened because Apple cannot check it for malicious software. This software needs to be updated. Contact the developer for more information.
> 
> <img alt="image" width="186" src="https://private-user-images.githubusercontent.com/151413186/309682898-30b248a4-a2ae-466b-90d9-4286c1f87f56.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1NDE4MTEsIm5iZiI6MTcwOTU0MTUxMSwicGF0aCI6Ii8xNTE0MTMxODYvMzA5NjgyODk4LTMwYjI0OGE0LWEyYWUtNDY2Yi05MGQ5LTQyODZjMWY4N2Y1Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNFQwODM4MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NzViYTliMDFjZjY1MTM1YTliMzc5YzQ5YjAyMGFkZmQzNjYxNDJiMDJhMjljYzY0MWNiODllNmVhNWRiNjQyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ywe9-b8JTEpT0IhWd4eYyRnut1Be1WJEHTprp5nAKk8">

Can you try dragging the app out from .dmg to another folder, and open it by right clicking it.
Probably signing issue since i don't have valid signing certificate

[image]
[image]

Then open

sidewinder666, 2024-03-04:
Radito, while that **did** work to open the app without a crash, the following was shown once it opened:

[image]

radito, 2024-03-04:
> Radito, while that **did** work to open the app without a crash, the following was shown once it opened:
> 
> ![Monosnap 2024-03-04 06-46-38](https://private-user-images.githubusercontent.com/11396091/309758322-d573f154-df66-4fc6-b132-d7d233bdf684.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1NjE5NDQsIm5iZiI6MTcwOTU2MTY0NCwicGF0aCI6Ii8xMTM5NjA5MS8zMDk3NTgzMjItZDU3M2YxNTQtZGY2Ni00ZmM2LWIxMzItZDdkMjMzYmRmNjg0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA0VDE0MTQwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZhZGZlZDk0ODBiMTkwYTNkZjI3ZGNjZDUxY2MyMzIxZmQ0ZjAxNDM1MGViNDE4NjUzZjRkYzg5YWVkN2IwNjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.izkBoSVlK8ivAsAFaJMZeS6kV-kLuj5YRcDo6fPUH4E)

Okay thanks, i'm taking a look into it.

mg1011, 2024-03-10:
> > Radito, while that **did** work to open the app without a crash, the following was shown once it opened:
> 
> Okay thanks, i'm taking a look into it.

Can confirm using RSP1A on Mac M1 Pro with latest OS.  Also have RTL-SDR V4 and HF+ Discovery to test with if needed.

radito, 2024-03-11:
CubicSDR build with added SoapySDR Bundle config
https://github.com/radito/CubicSDR/releases/tag/0.2.8-2

If there's a macOS error ignore it, just right click -> open 
(because it's not signed with apple developer)

Feedback would be useful to validate if the issue is fixed.

[image]
[image]

sidewinder666, 2024-03-11:
Perhaps it's "just me" but I get the same symptoms as before. Opening it shows this:

[image]

2021 16" Macbook Pro M1, Sonoma 14.4

radito, 2024-03-11:
@sidewinder666 

can you try

`brew install soapysdr soapyrtlsdr`

see if that helps ?

sidewinder666, 2024-03-11:
Thanks for the suggestion, I had hoped it would help, but...
(I also hadn't run Homebrew in ages, that took a long time to get done. Brew updated two pages worth of... things... and I had to first install command-line tools to get something else updated. Thankfully, it all seemed to work. )
The two extensions installed without error, however when I open the app I get the exact same error. I can dismiss the error but then no devices show up at all.

radito, 2024-03-11:
Just figure it out my soapy sdr isn't in this build directory
`SET(SOAPY_SDR_MOD_PATH "${CMAKE_INSTALL_PREFIX}/lib/SoapySDR/modules${SOAPY_SDR_ABI_VERSION}")`

Need some adjustment, hope it will be fixed.

sidewinder666, 2024-03-11:
Umm, apologies, but... are you asking me to run that command in shell, or just thinking out loud what you're going to do to make another build?

radito, 2024-03-11:
No, you're fine.
i'm just informing for another build.

radito, 2024-03-11:
yeah, i think that's what's causing the problem.

Working on a fix rn.

[image]

mg1011, 2024-03-11:
So i have 3 different SDR's.  one is the SDRPlay1a.  I have both API installed.. the 2.x and the 3.x.  Long way of saying that one of those installed Soapy along with it.  I think it was the 2.x API.  its installed here by default

/usr/local/bin/SoapySDRUtil

michagar@MICHAGAR-M-QN2M ~ % SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=2303006698
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255

I also have a RTL-SDR V4 and Airspy HF+ Discovery to test anything with if needed.

radito, 2024-03-11:
https://github.com/radito/CubicSDR/releases/tag/0.2.8-3

0.2.8-3 build
Add multiple sdr library and fix conflicting build path from homebrew.

sidewinder666, 2024-03-11:
BINGO!

I had to allow some exceptions thru my firewall (Little Snitch), but it does work fine now.

Thanks for persevering! :)

mg1011, 2024-03-11:
Here is what I see now.  Should I see SDRPLAY in the list?

SoapySDRUtil --info

######################################################

##     Soapy SDR -- the SDR abstraction library     ##

######################################################

Lib Version: v0.7.0-g69c16e98

API Version: v0.7.0

ABI Version: v0.7

Install root: /usr/local

Search path: /usr/local/lib/SoapySDR/modules0.7

Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrPlaySupport.so
(0.1.0-12c3db6)

Available factories... sdrplay
[image: Screenshot 2024-03-11 at 2.08.36 PM.png]

mg1011, 2024-03-11:
[image]

radito, 2024-03-11:
I think it might be the problem with my library, taking a look

```
SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[ERROR] SoapySDR::loadModule(/opt/homebrew/lib/SoapySDR/modules0.8/libsdrPlaySupport.so)
  dlopen() failed: dlopen(/opt/homebrew/lib/SoapySDR/modules0.8/libsdrPlaySupport.so, 0x0001): Library not loaded: libsdrplay_api.so.3.14
  Referenced from: <04011C79-DA40-3F10-8149-67BEF99C411C> /opt/homebrew/lib/SoapySDR/modules0.8/libsdrPlaySupport.so
  Reason: tried: 'libsdrplay_api.so.3.14' (no such file), '/System/Volumes/Preboot/Cryptexes/OSlibsdrplay_api.so.3.14' (no such file), 'libsdrplay_api.so.3.14' (no such file), '/Users/xxx/libsdrplay_api.so.3.14' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/xxx/libsdrplay_api.so.3.14' (no such file), '/Users/xxx/libsdrplay_api.so.3.14' (no such file)
  ```

radito, 2024-03-12:
@mg1011 

Can you try my latest build see if it's working, Added multiple library

[FORK] Include libsdrplay_api.so.3.14 and libaudioSupport.so
https://github.com/radito/CubicSDR/releases/tag/0.2.8-4

Thanks

mg1011, 2024-03-12:
> @mg1011
> 
> Can you try my latest build see if it's working, Added multiple library
> 
> [FORK] Include libsdrplay_api.so.3.14 and libaudioSupport.so https://github.com/radito/CubicSDR/releases/tag/0.2.8-4
> 
> Thanks

Wow, working well!  Rig Control/Tracking works and also noticed some more options showing up with this one.

<img width="706" alt="Screenshot 2024-03-12 at 8 25 56 AM" 
src="https://github.com/cjcliffe/CubicSDR/assets/56657381/e283a327-18ce-434d-83c0-297c5d4fe33e">
[image]

cts409, 2024-03-19:
Hello!
On my Air M1, the latest build v.0.2.8-4 starts, but the SDR is not displayed.
Tested on "POP MSI SDR NANO" and "POP MSI SDRPro".

[image]

mg1011, 2024-03-19:
Appears that that is a CLONE of the SDRPlay?  I wonder if its not working
because the drivers/api for SDRPlay don't work with the clone ?

radito, 2024-03-19:
> Hello! On my Air M1, the latest build v.0.2.8-4 starts, but the SDR is not displayed. Tested on "POP MSI SDR NANO" and "POP MSI SDRPro".

Can you try `SoapySDRUtil --probe` see if the device shows up and the driver recognized it ?

cts409, 2024-03-19:
macOS Ventura

> Appears that that is a CLONE of the SDRPlay? I wonder if its not working because the drivers/api for SDRPlay don't work with the clone ?
> […](#)

On macOS Ventura and arm64_CubicSDR-0.2.7-Darwin it was detected and worked. I tried downgrading from macOS Sonoma to macOS Ventura and everything started working.

cts409, 2024-03-19:
> > Hello! On my Air M1, the latest build v.0.2.8-4 starts, but the SDR is not displayed. Tested on "POP MSI SDR NANO" and "POP MSI SDRPro".
> 
> Can you try `SoapySDRUtil --probe` see if the device shows up and the driver recognized it ?

$ SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
Error probing device: No RTL-SDR devices found

mg1011, 2024-03-19:
Looks like the driver doesn't detect that SDR.  Has it worked on your Mac before ?

This is what the SDRPlay RSP1A (non clone) looks like when running the probe.

SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=2303006698
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 48] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 9] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

cts409, 2024-03-19:
Yes, it worked for macOS Ventura and Arm64_CubicSDR-0.2.7-Darwin. But perhaps I also installed the SDRplay_RSP_API-MacOSX-2.13.2.pkg driver, I don’t remember. I'll downgrade my MacBook and test it again.

cts409, 2024-03-19:
macOS Ventura 13.6.5 + CubicSDR v0.2.4

!!!!!!!  "POP MSI SDRPro" !!!!!!!
SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=null
  hardware=null

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 0 Rx, 0 Tx
  Timestamps: NO

!!!!!!! "POP MSI SDR NANO" !!!!!!!
SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=B0000P0000
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=1, type=string, options=(0, 1, 2, 3)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 42] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 3] dB
  Full freq range: [0.01, 2000] MHz
   RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

[image]

SpaceCadet1249, 2024-03-22:
Letter to GitHub team:
Hello there! I’m new here, but have a MacBook Pro M2 and a Yaesu FTDX1200 transceiver.
I’ve searched the web and YouTube to see what software offers rig-control that runs on my Mac.
A gentleman by the name of Steve (with SDRplay) has a 5 year old YouTube showing just how to set my rig up on a Mac with CubicSDR.
Imagine my disappointment when it wouldn’t run on my Mac. So I searched the web, hoping for a solution.
Boy was I happy when I ran into CJCLIFFE’S issues 1011 thread and downloaded CubicSDR version 0.2.8-4 
and IT WORKED! That is until I tried setting up Rig-Control. NOW THE SDRPLAY WILL NOT INITIALIZE ANYMORE.
I have tried both the initial “SDR Devices” window, and the File menu’s “Start / Stop Device” window. All I get is a multicolored Beach Ball for awhile, and it gives up.

I realize it’s probably because I did something stupid, but Let me explain:

The only USB to Serial cable I had ended in a Male DB9 connector. I had to use a gender-changer, and I have learned since then that when you do that, pins 2 and 3 are switched. 

I believe that cable confused CubicSDR and it wrote something in its config file that is preventing the startup of the SDRplay box now. Why do I think that? 
1. Because when I tried using rig control the Yaesu actually changed to some miscellaneous frequency and made some clicking noises. 
2. I know the SDRplay box wasn’t damaged in any way because the SDRconnect software still works fine.
3. Deleting and reinstalling the CubicSDR ver 0.2.8-4 did not resolve the problem. The SDRplay box will not startup so I get no radio reception at all now.

So, I just got a new USB-Serial cable made by Dtech and running the FTDI chip. I have their driver that is specific for 
ARM-based Apple Silicon architecture (M1, M2, M3 chips).
So I’m ready to test RIG-Control again.

If only I could get the SDRplay box started again…

Possible Solution:
If I could find the config file and delete it, I’ll bet everything would work again. 

Where is it?

Oh, here’s a screenshot of how it looks now, minus the beach ball. It doesn’t get captured for some reason.

So, can you tell me where the config file is so I can delete it?
I’ll bet that’ll fix the problem.

[image]

SpaceCadet1249, 2024-03-23:
OK. I found the Config file, deleted it, and it all works.
/Users/username/Library/Application Support/CubicSDR
Filename: config.xml
Just can't get rig-control to work, but that's probably on another thread.
Thanks for this great application!

radito, 2024-03-23:
@SpaceCadet1249  The build uses the same code as the repo, there's nothing changed.
Are you able to try the old version see if the Rig Control works in your specific setup ? If not then it's not supported by cubic sdr and probably need support from the maintainer. And probably you need to do some troubleshooting first like debugging what's coming out from serial and try with different baud rate see if that's help.

And it appears @mg1011 can use rig control and he has no problem.

radito, 2024-03-23:
@cts409 

For the build i'm including the library from the official SDRPlay website the version is [SDRplayAPI-macos-installer-universal-3.14.0.pkg](https://www.sdrplay.com/software/SDRplayAPI-macos-installer-universal-3.14.0.pkg)

From your case it seems the soapySDR hasn't detected the SDR and probably the the clone is not supported by this API ? I'm not so sure can you try this specific api version in sonoma and do the probe again ?

SpaceCadet1249, 2024-03-23:
> @SpaceCadet1249 The build uses the same code as the repo, there's nothing changed. Are you able to try the old version see if the Rig Control works in your specific setup ? If not then it's not supported by cubic sdr and probably need support from the maintainer. And probably you need to do some troubleshooting first like debugging what's coming out from serial and try with different baud rate see if that's help.
> 
> And it appears @mg1011 can use rig control and has no problem.

@radito 
This is the first build that I could get to run on my MacBook M2, since all the others crashed on startup.
I believe there's some kind of problem with the SoapySDR library since its reception is very weak compared to SDRconnect. For instance WWV on 15mhz tonight (on the Yaesu) was 10 points above S9, yet I could not even hear it using CubicSDR.
And every time I start the CubicSDR I must DELET the config file in order to start the SDRplay box. It's not a counterfeit, but the real RSP1b device.
I'm sorry, but I'm not a programmer and wouldn't know how to debug the serial port. I'm using 9600 baud N81. Matching settings on my Yaesu FTDX-1200. 
If I should be looking at another thread for this, please point me to it. thanks

cts409, 2024-03-23:
> @cts409
>
> For the build i'm including the library from the official SDRPlay website the version is [...]

Works

SpaceCadet1249, 2024-03-23:
@cts409 If you're interested, here's a short video of SDRplay trying to initialize. Bear in mind it tries to come up with the RF frequency set to zero. I have tried setting it to, say 15Mhz before initializing, but get the same results. To exit the program at this point, I have to do a Forced quit. The only way I can get it to initialize is to blow away the config.xml file AND unplug and replug the SDRplay USB cable.
Then it can initialize again. 
First is a picture where you can see the SDRplay is actually detected. 
Then next there's the video. If you unmute the speaker you can hear my narration.

[image]

https://github.com/cjcliffe/CubicSDR/assets/164538664/9d4bc062-5940-4101-924b-6f1311c84cec

SpaceCadet1249, 2024-03-31:
@cts409 @radito @mg1011
When I finally found a USB > UART cable that worked, all these other problems went away.
For details on which two cables I've found that work, and why, please see #1028 
I've been using CubicSDR version  0.2.8-4-Darwin-Arm64 RELIABLY for over a week now. Thanks!

SteveClement, 2024-05-21:
> @sidewinder666
> 
> can you try
> 
> `brew install soapysdr soapyrtlsdr`
> 
> see if that helps ?

This solved the issue for me.

jforge, 2024-05-23:
@radito Thanks for all the effort to make something available for ARM64.

I've followed so far on a Apple M3 Pro with Sonoma 14.5, but had no final success. Is there some further advice?

- The `CubicSDR.0.2.8-aarch64.dmg` works using `Open` manually as described
- I first had no success with `SoapySDRUtil --probe` but doing `brew install soapysdr soapyrtlsdr` lists my device (I have a `Nooelec NESDR SMArTee v2`:
```
SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device
Found Rafael Micro R820T tuner
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Found Rafael Micro R820T tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=R820T
  index=0
  origin=https://github.com/pothosware/SoapyRTLSDR

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: YES
  Time sources: sw_ticks
  Other Settings:
     * Direct Sampling - RTL-SDR Direct Sampling Mode
       [key=direct_samp, default=0, type=string, options=(0, 1, 2)]
     * Offset Tune - RTL-SDR Offset Tuning Mode
       [key=offset_tune, default=false, type=bool]
     * I/Q Swap - RTL-SDR I/Q Swap Mode
       [key=iq_swap, default=false, type=bool]
     * Digital AGC - RTL-SDR digital AGC Mode
       [key=digital_agc, default=false, type=bool]
     * Bias Tee - RTL-SDR Blog V.3 Bias-Tee Mode
       [key=biastee, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Size - Number of bytes per buffer, multiples of 512 only.
       [key=bufflen, units=bytes, default=262144, type=int]
     * Ring buffers - Number of buffers in the ring.
       [key=buffers, units=buffers, default=15, type=int]
     * Async buffers - Number of async usb buffers (advanced).
       [key=asyncBuffs, units=buffers, default=0, type=int]
  Antennas: RX
  Full gain range: [0, 49.6] dB
    TUNER gain range: [0, 49.6] dB
  Full freq range: [23.999, 1764] MHz
    RF freq range: [24, 1764] MHz
    CORR freq range: [-0.001, 0.001] MHz
  Sample rates: [0.225001, 0.3], [0.900001, 3.2] MSps
  Filter bandwidths: [0, 8] MHz
```
- Still, when Opening the application, there is no suche SDR device
[image]

radito, 2024-05-29:
@jforge are you using the latest release ?

0.2.8-4 [Latest](https://github.com/radito/CubicSDR/releases/latest)

jforge, 2024-05-29:
@radito Thanks for the hint, you're right, I grabbed the `0.2.8-aarch64` from elsewhere.

With `0.2.8-4` everything looks fine from my side.

[image]

SpaceCadet1249, 2024-05-29:
Hello:

That version is working fine for me too.

Thanks!

Vic

Screenshot.2024-05-29.at.11.00.28.png (view on web)<https://github.com/cjcliffe/CubicSDR/assets/307481/92675d46-a4df-432f-a611-efa1ad0e468c>

closed 2026-11-15 by cjcliffe