cjcliffe/CubicSDR #1015: No HACKRF One Support

shull895, 2023-10-28:
I recently installed CubicSDR ver 0.2.5 and discovered it does not support my HackRF One any longer. It says it has built in SoapySDR support integrated into the software BUT it is not recognizing my Hackrf One in Windows 11. Other SDR software like SDR++ and SDR# both work. If you state it supports my SDR then make sure it does. L:ies are not helpfull.

dmaltsiniotis, 2023-10-28:
@shull895 Unfounded accusations are also not helpful. If you state there is a problem, then make sure to provide the relevant details of your environment so that folks might be able to help you. CubicSDR has and does support HackRF One, as I've been using it for _years._

luzpaz, 2023-11-17:
Close issue ?

dmaltsiniotis, 2024-01-31:
I had a similar problem when there is a mismatch between the Soapy library that was bundled with CubicSDR, and one that I had on my system path. Is CubicSDR perhaps finding a different version of the Soapy library on your system that is not compatible with the version it was built with?

The `Available factories...null` is somewhat suspect though.

Yvanttt, 2024-02-02:
success step 1,
did a yum erase CubicSDR, it removed distro version of CubicSDR, Liquid-dsp and rtaudio.

but i get no audio:

running as root"

[INFO] Opening HackRF One #0 88869dc354b561b...
SDR thread starting.
device init()
SDRThread : Device Stream set to MTU: 131072
starting readLoop()

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

No audio devices found!

as user: 
/usr/bin/CubicSDR 
Loaded 283 rig models via hamlib.

Audio Device #0 CM108 Audio Controller
        Default Output? Yes
        Default Input? Yes
        Input channels: 1
        Output channels: 2
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                24-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                8000hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                96000hz
                192000hz

Audio Device #1 USB 2.0 Camera
        Default Output? No
        Default Input? No
        Input channels: 2
        Output channels: 0
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                24-bit signed integer.
                32-bit signed integer.
                32-bit float normalized between plus/minus 1.0.
        Supported sample rates:
                8000hz
                16000hz
                22050hz
                32000hz
                44100hz
                48000hz
                96000hz
                192000hz

Segmentation fault (core dumped)

reinstalled via "yum install Liquid-dsp and rtaudio", still no audio.
removed distro versio and install the one compiled from source, still no aidio