cjcliffe/CubicSDR #1024: Waterfall goes dark when I zoom in CubicSDR.

tteletubby, 2024-02-12:
[image]

My sample rate is: 1.920 MHz
CPU Usage: High + enhanced DSP
SDR: RTL-SDR Blog V3

What's going on?

dmaltsiniotis, 2024-02-13:
What kind of hardware are you running this on?

tteletubby, 2024-02-13:
**Debian 12 + KDE
Model name      : Intel(R) Core(TM) i3-4000M CPU 
CPU MHz         : 1596.452
Processor: 3**

egeste, 2024-04-28:
+1

```sh
$ uname -a
Linux debian 6.1.0-18-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.76-1 (2024-02-01) x86_64 GNU/Linux

$ hwinfo --cpu --short
cpu:                                                            
                       Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, 3500 MHz
                       Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, 3406 MHz
                       Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, 3500 MHz
                       Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, 3485 MHz
                       Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, 3485 MHz
                       Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, 3492 MHz
                       Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, 3411 MHz
                       Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, 3400 MHz
```

whudson, 2024-07-06:
I have the same issue.

HackRF One
Debian Sid, 6.7.12-amd64, GNOME
Intel(R) Core(TM) i5-4690K CPU @ 3.50GHz

whudson, 2024-07-06:
There is a thread on this website from someone having the same problem that says Fedora works OK.

https://www.reddit.com/r/sdr/comments/yxglrs/cubicsdr_help_modem_spectrum_and_waterfall/

whudson, 2024-07-06:
I just built Cubic from source CubicSDR@0333303 to try and debug and I'm not experiencing the issue on my built version.

Debian is still shipping v0.2.7

Seems like this is not an issue with this repo.

closed 2026-11-15 by cjcliffe