cjcliffe/CubicSDR #1036: CubicSDR crash at startup on MacOS Sonoma 14.6.1 Intel

alfonsofiorentino, 2024-08-19:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               CubicSDR [4582]
Path:                  /Applications/CubicSDR.app/Contents/MacOS/CubicSDR
Identifier:            com.cubicproductions.cubicsdr
Version:               0.2.4 (0.2.4)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-08-19 12:41:07.1449 +0200
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Bridge OS Version:     8.6 (21P6074)
Anonymous UUID:        03EE4382-FDDD-B246-1204-58913821F2EE

Time Awake Since Boot: 12000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       UNKNOWN_0xD at 0x0000000000000000
Exception Codes:       0x000000000000000d, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [4582]

VM Region Info: 0 is not in any region.  Bytes before following region: 4543393792
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10eceb000-10f25a000    [ 5564K] r-x/rwx SM=COW  /Applications/CubicSDR.app/Contents/MacOS/CubicSDR

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   CubicSDR                      	       0x10ef4c880 wxMacCoreGraphicsBrushData::CalculateShadingValues(void*, double const*, double*) + 240
1   CoreGraphics                  	    0x7ff81bc76b68 CGFunctionEvaluate + 243
2   CoreGraphics                  	    0x7ff81bc76912 function_evaluate + 467
3   CoreGraphics                  	    0x7ff81bc75ff3 ripc_AcquireFunction + 844
4   CoreGraphics                  	    0x7ff81bc75307 ripc_DrawShading + 5134
5   CoreGraphics                  	    0x7ff81c0ba4c9 CG::DisplayListExecutor::drawShading(CG::DisplayListEntryShading const*) + 435
6   CoreGraphics                  	    0x7ff81c1a4d3b CG::DisplayList::executeEntries(std::__1::__wrap_iter<std::__1::shared_ptr<CG::DisplayListEntry const>*>, std::__1::__wrap_iter<std::__1::shared_ptr<CG::DisplayListEntry const>*>, CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*, bool) + 195
7   CoreGraphics                  	    0x7ff81bcf9631 CG::DisplayList::execute(CGContextDelegate*, CGRenderingState*, CGGStack*, CGRect const*, __CFDictionary const*) + 355
8   CoreGraphics                  	    0x7ff81bcf93e0 CGDisplayListDrawInContextDelegate + 558
9   AppKit                        	    0x7ff819cb6fc9 0x7ff81981a000 + 4837321
10  AppKit                        	    0x7ff819c98439 0x7ff81981a000 + 4711481
11  QuartzCore                    	    0x7ff81e7f233f CABackingStoreUpdate_ + 630
12  QuartzCore                    	    0x7ff81e851076 invocation function for block in CA::Layer::display_() + 53
13  QuartzCore                    	    0x7ff81e7f13cc -[CALayer _display] + 2251
14  AppKit                        	    0x7ff819c98304 0x7ff81981a000 + 4711172
15  AppKit                        	    0x7ff819c97d81 0x7ff81981a000 + 4709761
16  AppKit                        	    0x7ff819c983c5 0x7ff81981a000 + 4711365
17  QuartzCore                    	    0x7ff81e7f0361 CA::Layer::display_if_needed(CA::Transaction*) + 867
18  QuartzCore                    	    0x7ff81e988400 CA::Context::commit_transaction(CA::Transaction*, double, double*) + 758
19  QuartzCore                    	    0x7ff81e7d11a9 CA::Transaction::commit() + 725
20  AppKit                        	    0x7ff8199b5a7d __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285
21  AppKit                        	    0x7ff81a4a6854 ___NSRunLoopObserverCreateWithHandler_block_invoke + 41
22  CoreFoundation                	    0x7ff8161ea586 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
23  CoreFoundation                	    0x7ff8161ea4aa __CFRunLoopDoObservers + 493
24  CoreFoundation                	    0x7ff8161e9a2c __CFRunLoopRun + 850
25  CoreFoundation                	    0x7ff8161e9112 CFRunLoopRunSpecific + 557
26  HIToolbox                     	    0x7ff820c08a09 RunCurrentEventLoopInMode + 292
27  HIToolbox                     	    0x7ff820c08646 ReceiveNextEventCommon + 201
28  HIToolbox                     	    0x7ff820c08561 _BlockUntilNextEventMatchingListInModeWithFilter + 66
29  AppKit                        	    0x7ff819859171 _DPSNextEvent + 880
30  AppKit                        	    0x7ff81a16daf0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1273
31  AppKit                        	    0x7ff81984a585 -[NSApplication run] + 603
[... 274 lines cut ...]
TOTAL                              2.7G     1665 

-----------
Full Report
-----------

{"app_name":"CubicSDR","timestamp":"2024-08-19 12:41:07.00 +0200","app_version":"0.2.4","slice_uuid":"26287744-3b0b-359a-9613-0649e726d37a","build_version":"0.2.4","platform":1,"bundleID":"com.cubicproductions.cubicsdr","share_with_app_devs":1,"is_first_party":0,"bug_type":"309","os_version":"macOS 14.6.1 (23G93)","roots_installed":0,"name":"CubicSDR","incident_id":"D2AC2D8A-0957-410D-8DAC-3E44F61B4EDA"}
{
  "uptime" : 12000,
  "procRole" : "Foreground",
  "version" : 2,
  "userID" : 501,
  "deployVersion" : 210,
  "modelCode" : "MacBookPro16,1",
  "coalitionID" : 3072,
  "osVersion" : {
    "train" : "macOS 14.6.1",
    "build" : "23G93",
    "releaseType" : "User"
  },
  "captureTime" : "2024-08-19 12:41:07.1449 +0200",
  "codeSigningMonitor" : 0,
  "incident" : "D2AC2D8A-0957-410D-8DAC-3E44F61B4EDA",
  "pid" : 4582,
  "cpuType" : "X86-64",

tibbon, 2024-09-10:
I'm seeing similar. I tried it from the console and the last line was `segmentation fault  /Applications/CubicSDR.app/Contents/MacOS/CubicSDR`

jminor, 2024-12-29:
Same crash at startup on macOS Sequoia 15.1.1 running CubicSDR 0.2.5

BasilBibi, 2025-02-11:
Nice one @jminor thank you! I got 0.2.3 running on 14.3.1