cjcliffe/CubicSDR #1038: CubicSDR not compatible with rtaudio 6.0.1

hobbes1069, 2024-10-19:
I had a request to update rtaudio on Fedora but did not catch that CubicSDR was not compatible until after the build.

``
/builddir/build/BUILD/CubicSDR-0.2.7-build/CubicSDR-0.2.7/src/audio/AudioThread.cpp: In member function ‘void AudioThread::setupDevice(int)’:
/builddir/build/BUILD/CubicSDR-0.2.7-build/CubicSDR-0.2.7/src/audio/AudioThread.cpp:435:12: error: ‘RtAudioError’ does not name a type; did you mean ‘RtAudioErrorType’?
  435 |     catch (RtAudioError& e) {
      |            ^~~~~~~~~~~~
      |            RtAudioErrorType
/builddir/build/BUILD/CubicSDR-0.2.7-build/CubicSDR-0.2.7/src/audio/AudioThread.cpp:436:9: error: ‘e’ was not declared in this scope
  436 |         e.printMessage();
      |         ^
/builddir/build/BUILD/CubicSDR-0.2.7-build/CubicSDR-0.2.7/src/audio/AudioThread.cpp: In member function ‘virtual void AudioThread::run()’:
/builddir/build/BUILD/CubicSDR-0.2.7-build/CubicSDR-0.2.7/src/audio/AudioThread.cpp:539:16: error: ‘RtAudioError’ does not name a type; did you mean ‘RtAudioErrorType’?
  539 |         catch (RtAudioError& e) {
      |                ^~~~~~~~~~~~
      |                RtAudioErrorType
/builddir/build/BUILD/CubicSDR-0.2.7-build/CubicSDR-0.2.7/src/audio/AudioThread.cpp:540:13: error: ‘e’ was not declared in this scope
  540 |             e.printMessage();
      |             ^
```

Full build log (until it expires):
https://kojipkgs.fedoraproject.org//work/tasks/5675/124965675/build.log

dreua, 2025-04-16:
@cjcliffe would you accept a PR and do a new release if someone fixed this?

ohhai, 2025-04-20:
Just in case: seems like a similar fix, not compatible with v5 though:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1051556
https://salsa.debian.org/debian-hamradio-team/soapysdr/soapyaudio/-/commit/499e87f20942754050e68e61b0d370c53d34bdcd

This change seems to be compatible with both versions 5 and 6:

https://github.com/AlexandreRouma/SDRPlusPlus/issues/1149#issuecomment-1902845673
https://github.com/AlexandreRouma/SDRPlusPlus/pull/1284

cjcliffe, 2026-11-15:
Thanks for the report -- master branch and builds now use RTAudio 6.0.1

closed 2026-11-15 by cjcliffe