cjcliffe/CubicSDR #1042: CubicSDR does not compile with Hamlib 4.6

srs4511351, 2024-12-26:
Hamlib 4.6 has been released. make fails with errors.
When built without hamlib, the build is successful and CubicSDR runs.

I have a Raspberry Pi 5 with Dietpi bookworm OS.

Can you make it work with either hamlib 4.5.5 or 4.6?

[CubicSDR cmake build log hamlib 4.6.txt](https://github.com/user-attachments/files/18255339/CubicSDR.cmake.build.log.hamlib.4.6.txt)

srs4511351, 2025-01-02:
Hamlib has recently reverted the change that broke the CubicSDR build with this comment:
`Change rig_list_foreach back to using const argument -- was breaking many C++ application builds`

I can now build it with Hamlib 4.7~git

cjcliffe, 2026-11-15:
Thanks for the report -- this is now patched in master to work with all 4.5 -> 4.7

closed 2026-11-15 by cjcliffe