cjcliffe/CubicSDR #1043: AppImage does not containe some libs

jpVm5jYYRE1VIKL, 2024-12-31:
./CubicSDR-0.2.5-x86_64.AppImage
./CubicSDR-0.2.5-x86_64.AppImage: symbol lookup error: /usr/lib/libgio-2.0.so.0: undefined symbol: g_module_open_full

joostvhts, 2025-02-11:
for me it's these:

```
./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libthai.so.0: version `LIBTHAI_0.1.25' not found (required by /lib/x86_64-linux-gnu/libpango-1.0.so.0)
./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib/x86_64-linux-gnu/libgio-2.0.so.0)
./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib/x86_64-linux-gnu/libmount.so.1)
```

angelosne, 2025-02-14:
I had this error

```
./CubicSDR-0.2.4-x86_64.AppImage: error while loading shared libraries: libnsl.so.1: cannot open shared object file: No such file or directory
``` 
and after installing the libnsl I got this one

```
./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libthai.so.0: version `LIBTHAI_0.1.25' not found (required by /lib64/libpango-1.0.so.0)
./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib64/libgio-2.0.so.0)
./CubicSDR-0.2.4-x86_64.AppImage: ./lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib64/libmount.so.1)
``` 
I use Fedora 41

ominty, 2025-04-25:
Debian 12 (bookworm) Cinnamon and Gnome desktops:
`./CubicSDR-0.2.5-x86_64.AppImage: ./lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib/x86_64-linux-gnu/libgio-2.0.so.0)
./CubicSDR-0.2.5-x86_64.AppImage: ./lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /lib/x86_64-linux-gnu/libmount.so.1)
./CubicSDR-0.2.5-x86_64.AppImage: symbol lookup error: /lib/x86_64-linux-gnu/libgio-2.0.so.0: undefined symbol: g_module_open_full
`

cjcliffe, 2026-11-15:
New AppImages for 0.3.0 will build on newer base; closing this one for now

closed 2026-11-15 by cjcliffe