cjcliffe/CubicSDR #1047: App crash apple macbook pro M1 2021 Version

tamastom, 2025-04-30:
After start the app crashes, first after installation. Reinstall does not work either.

[crashreport.txt](https://github.com/user-attachments/files/19972419/crashreport.txt)

suvl, 2025-05-02:
+1, happening here too

actualben, 2025-05-15:
I'm also getting a segfault on an arm Mac. It starts OK, but when the window gets focus, it segfaults.

Running it with lldb shows:

```
SDR enumerator done.
Process 91655 stopped
* thread #18, queue = 'CA::CG::Queue', stop reason = EXC_BAD_ACCESS (code=1, address=0x40449fa000000008)
    frame #0: 0x0000000100261880 CubicSDR`wxMacCoreGraphicsBrushData::CalculateShadingValues(void*, double const*, double*) + 240
CubicSDR`wxMacCoreGraphicsBrushData::CalculateShadingValues:
->  0x100261880 <+240>: movq   0x8(%rax), %rcx
    0x100261884 <+244>: movq   %rcx, (%rdx)
    0x100261887 <+247>: movq   0x10(%rax), %rcx
    0x10026188b <+251>: movq   %rcx, 0x8(%rdx)
Target 0: (CubicSDR) stopped.

(lldb) bt
* thread #18, queue = 'CA::CG::Queue', stop reason = EXC_BAD_ACCESS (code=1, address=0x40449fa000000008)
  * frame #0: 0x0000000100261880 CubicSDR`wxMacCoreGraphicsBrushData::CalculateShadingValues(void*, double const*, double*) + 240
    frame #1: 0x00007ff8207f12ff CoreGraphics`CGFunctionEvaluate + 243
    frame #2: 0x00007ff8241208cc QuartzCore`CA::CG::(anonymous namespace)::FunctionSampler::sample(int) + 150
    frame #3: 0x00007ff8241207bf QuartzCore`CA::CG::(anonymous namespace)::FunctionColormapGenerator::generate(CA::CG::Renderer&, CGColorSpace*, unsigned char*, int, bool) const + 221
    frame #4: 0x00007ff823e83cba QuartzCore`CA::OGL::Mosaic::draw(CA::OGL::Context&, unsigned int, unsigned int, CA::OGL::Mosaic::Key const*, int, int, bool, CA::OGL::Mosaic::Delegate const&, unsigned int, CA::Mat2<double> const*) + 1478
    frame #5: 0x00007ff82411ef10 QuartzCore`CA::CG::(anonymous namespace)::draw_shading_delegate(CA::CG::(anonymous namespace)::ShadingDelegate&, unsigned int, unsigned int) + 162
    frame #6: 0x00007ff823ed003c QuartzCore`CA::CG::fill_shading(CA::CG::Renderer&, CGShading*, CA::Mat2<double> const&, bool) + 1397
    frame #7: 0x00007ff823e6ba69 QuartzCore`CA::CG::DrawOp::render(CA::CG::Renderer&) const + 4059
    frame #8: 0x00007ff823e67f50 QuartzCore`CA::CG::Queue::render_callback(void*) + 1332
    frame #9: 0x00007ff81a63b9fc libdispatch.dylib`_dispatch_client_callout + 6
    frame #10: 0x00007ff81a62c468 libdispatch.dylib`_dispatch_lane_serial_drain + 1044
    frame #11: 0x00007ff81a62cdad libdispatch.dylib`_dispatch_lane_invoke + 382
    frame #12: 0x00007ff81a635b96 libdispatch.dylib`_dispatch_root_queue_drain_deferred_wlh + 275
    frame #13: 0x00007ff81a6354e3 libdispatch.dylib`_dispatch_workloop_worker_thread + 688
    frame #14: 0x00007ff81a7cd8ad libsystem_pthread.dylib`_pthread_wqthread + 298
    frame #15: 0x00007ff81a7cc843 libsystem_pthread.dylib`start_wqthread + 15

```

Possibly related to: https://github.com/wxWidgets/wxWidgets/issues/24116

Want a PR bumping wxWidgets from 3.2.1 to 3.2.8?

actualben, 2025-05-15:
oops, looks like this duplicates issue #1036

marklacas, 2025-06-30:
Yep, same here.  M1 crash.

IgnacioHR, 2025-08-11:
Same here. Can't start app on my macbook pro